QuinnDamerell/OctoPrint-OctoEverywhere #1: software update hook is incorrectly formed

jneilliii, 2020-11-07:
https://github.com/QuinnDamerell/OctoPrint-OctoEverywhere/blob/b00848fde6849da74480a6c956d99b90da2eb416/octoprint_octoeverywhere/__init__.py#L66

This line needs to return a dictionary that matches your plugin's identifier, case sensitive.

So should be `octoeverywhere=dict(`

QuinnDamerell, 2020-11-07:
Thanks so much for looking and catching that! Fixed now.

I'm new to python, so feel free to leave any other comments or issues you see! I have quite thoroughly tested the plugin -> website logic, but the OctoPrint integration I did as well as I could.

QuinnDamerell, 2020-11-07:
Link to fix: https://github.com/QuinnDamerell/OctoPrint-OctoEverywhere/commit/3e387515e0e9e559e77d3b47873f4913caac1dbe

jneilliii, 2020-11-07:
Don't forget to publish a release that matches your version in setup.py here in your repo. That way when you release an update the software update check will work properly.

QuinnDamerell, 2020-11-07:
Oh thanks for that! I was under the impression that the update system pulled the master archive whenever it's newer. 

So to update, I need to bump the version in setup.py, push, and make a github release matching the same version?

jneilliii, 2020-11-08:
Exactly. The initial install will always pull from the install url provided in the plugin repo, which in most cases is the master.zip download, but updates from then on are all handled between the end-user's octoprint instance and github directly.

jneilliii, 2020-11-08:
I just noticed that the tag you used on the release was 0.0.10. Should be 0.1.0 based on your setup.py. It's the tag that is the important part that has to match I'm pretty sure.

QuinnDamerell, 2020-11-10:
@jneilliii sorry to bother you in this thread, but I have a bug where user's can close the wizard dialog and I'm at a loss on how to fix it. I found this in the docs

> Note that with the current implementation, all wizard dialogs will be will always be sorted by their mandatory attribute (which defaults to False) and then alphabetically by their name. Hence, mandatory wizard steps will come first, sorted alphabetically, then the optional steps will follow, also alphabetically. 

But I can't for the life of me figure out where to set `mandatory`. I don't want it set, and I think that's the only thing that can be causing this issue. Any ideas?

closed 2020-11-10 by jneilliii