QuinnDamerell/OctoPrint-OctoEverywhere #2: unable to register

jneilliii, 2020-11-07:
When I go to https://octoeverywhere.com/ I get a title, site under construction in the browser and that's it.

jneilliii, 2020-11-07:
oh, looks like it's back. I'm guessing the site is getting barraged since the notice just went out.

closed 2020-11-07 by jneilliii

QuinnDamerell, 2020-11-07:
Yeah! My deployment system also have a problem about 20% of the time where it fails and I have to kick it again. :)

Btw, I would love to hear your feedback on the system. I think you will quickly figure out how it works (proxying through the website with a websocket tunnel). There are a few tweaks I have in mind I was going to suggest (or attempt to make) for OctoPrint if I find time. For example, the "files" download links aren't relative urls, so they fail when clicked in OctoEverywhere.