QuinnDamerell/OctoPrint-OctoEverywhere #4: Can't connect to printer

Kawuezel, 2020-11-09:
Hi,

I successfully integrated one of my printers to OctoEverywhere but when I click on "Connect" I'm getting the following error after a while:

Your printer is connected, but the request took too long. This usually means OctoPrint is unresponsive or there is an intense process running on the printer. Try to access it locally if you can or you might have to start it.

Actually, OctoPrint is connected to my printer, printer is on and there is no print running. On the website, my printer is marked as "Online".

Thanks,
Kim

QuinnDamerell, 2020-11-10:
Thanks for the feedback! This is mostly likely due to the server issues we were having, but I think they have all been sorted out now!

Try again and let me know if it works. :)

Kawuezel, 2020-11-10:
Unfortunately, I have still the same issue.

QuinnDamerell, 2020-11-10:
@Kawuezel and @krudoy - can you try to update to the latest plugin (force it or just reinstall it) and try again? 

Do you know if your OctoPrint doesn't run on the default port? I fixed an issue with that in the plugin.

krudoy, 2020-11-11:
No luck/ I'm on 1.5 maintenance release

QuinnDamerell, 2020-11-11:
Do you know how to capture logs on the client side? Can you collect them and dm them to me on reddit? /u/quinbd

QuinnDamerell, 2020-11-11:
Also, do you know if your running py3 OctoPrint?

QuinnDamerell, 2020-12-20:
Hey @krudoy ! If you can still repo this please feel free to reopen. I have fixed a ton of things in the last few months, so I would bet it's fixed now. 👍

closed 2020-12-20 by QuinnDamerell