QuinnDamerell/OctoPrint-OctoEverywhere #5: Model dialog with no Next button after install

krudoy, 2020-11-10:
Modal Dialog Setup Wizard blocks everything.
Only 2 buttons [OctoEverywhere] which does nothing and [Previous] which is grayed out. No way to close/cancel/next. The only way to fix - uninstall plugin via SSH :(
[image]

QuinnDamerell, 2020-11-10:
I might have fixed this, can you try installing it again and seeing if the issue is resolved?

What version of octoprint are you using?

krudoy, 2020-11-10:
1.5

krudoy, 2020-11-10:
Installed with no luck.

Here is a list of plugins installed:

Package                          Version
-------------------------------- -------------
Arc-Welder                       1.0.1rc1.dev1
AstroPrint                       1.6.2
Babel                            2.8.0
backoff                          1.10.0
backports.csv                    1.0.7
backports.time-perf-counter      0.0.4
Backup-Scheduler                 0.0.4
Bed-Visualizer                   0.1.15
blinker                          1.4
bson                             0.5.10
cachelib                         0.1.1
certifi                          2020.6.20
chardet                          3.0.4
click                            7.1.2
colorzero                        1.1
DisplayLayerProgress             1.24.0
dropbox                          10.7.0
emoji                            0.6.0
feedparser                       6.0.2
filetype                         1.0.7
Flask                            1.1.2
Flask-Assets                     2.0
Flask-Babel                      1.0.0
Flask-Login                      0.5.0
Flask-Principal                  0.4.0
frozendict                       1.2
future                           0.18.2
Heatshrink                       0.3.2
heatshrink2                      0.9.0
idna                             2.10
ifaddr                           0.1.7
importlib-metadata               2.0.0
itsdangerous                     1.1.0
Jinja2                           2.11.2
jsonpickle                       1.4.1
Markdown                         3.1.1
MarkupSafe                       1.1.1
marlin-binary-protocol           0.0.7
Marlin-GCode-Documentation       0.11.0
netaddr                          0.8.0
netifaces                        0.10.9
numpy                            1.19.2
octoeverywhere                   0.2.2
OctoPrint                        0+unknown <-- this is how v 1.5 in maintenance branch appears 
OctoPrint-Dashboard              1.16.2
OctoPrint-Display-ETA            1.1.1
OctoPrint-Dropbox-Timelapse      0.1.4
OctoPrint-Enclosure              4.13.1
OctoPrint-ExcludeRegionPlugin    0.2.1.dev0
OctoPrint-FilamentManager        1.6.3
OctoPrint-FileCheck              2020.8.7
OctoPrint-FirmwareCheck          2020.9.23
OctoPrint-Fullscreen             0.0.5
OctoPrint-GcodeEditor            0.2.9
OctoPrint-Gpioshutdown           1.0.2
OctoPrint-MarlinBft              0.1.4
OctoPrint-MultipleUpload         1.1.0
OctoPrint-NavbarTemp             0.14
OctoPrint-PrettyGCode            1.2.2
OctoPrint-PrintTimeGenius        2.2.6
OctoPrint-Resource-Monitor       0.2.6
OctoPrint-TabOrder               0.5.11
OctoPrint-UltimakerFormatPackage 0.2.0
OctoPrint-WideScreen             0.1.3
OctoPrint-WS281x-LED-Status      0.6.0
pathtools                        0.1.2
picamerab                        1.13b1
Pillow                           8.0.1
pip                              20.2.4
pkg-resources                    0.0.0
pkginfo                          1.6.0
Preheat                          0.5.1
PrusaSlicer-Thumbnails           0.1.2
psutil                           5.7.2
pyasn1                           0.4.8
pylru                            1.2.0
pyserial                         3.4
python-dateutil                  2.8.1
pytz                             2020.1
PyYAML                           5.3.1
raven                            6.10.0
regex                            2020.10.23
requests                         2.24.0
requests-toolbelt                0.8.0
rpi-ws281x                       4.2.4
RPi.GPIO                         0.7.0
rsa                              4.0
sarge                            0.1.5.post0
semantic-version                 2.8.5
sentry-sdk                       0.19.1
setuptools                       50.3.2
sgmllib3k                        1.0.0
six                              1.15.0
SQLAlchemy                       1.1.18
STL-Viewer                       0.4.2
Themeify                         1.2.2
TheSpaghettiDetective            1.4.3
tornado                          5.1.1
Unidecode                        0.4.21
uritools                         2.1.1
urllib3                          1.25.11
watchdog                         0.10.3
webassets                        2.0
websocket-client                 0.57.0
Werkzeug                         1.0.1
wheel                            0.35.1
wrapt                            1.12.1
ws4py                            0.3.4
zeroconf                         0.24.5
zipp                             3.4.0

QuinnDamerell, 2020-11-10:
Thanks. I wonder if it's a bug in 1.5, I haven't tried it. For now I disabled the wizard for everyone, so it should prevent this until I have time to circle back around to it.

Thanks for your help and for reporting it!

closed 2020-11-10 by QuinnDamerell