QuinnDamerell/OctoPrint-OctoEverywhere #6: user's can close the wizard dialog

jneilliii, 2020-11-10:
@jneilliii sorry to bother you in this thread, but I have a bug where  and I'm at a loss on how to fix it. I found this in the docs

> Note that with the current implementation, all wizard dialogs will be will always be sorted by their mandatory attribute (which defaults to False) and then alphabetically by their name. Hence, mandatory wizard steps will come first, sorted alphabetically, then the optional steps will follow, also alphabetically. 

But I can't for the life of me figure out where to set `mandatory`. I don't want it set, and I think that's the only thing that can be causing this issue. Any ideas?

_Originally posted by @QuinnDamerell in https://github.com/QuinnDamerell/OctoPrint-OctoEverywhere/issues/1#issuecomment-724472235_

I just want to make sure I understand. They are able to click the close button and as a result the settings doesn't get saved [here](https://github.com/QuinnDamerell/OctoPrint-OctoEverywhere/blob/90a10c2d0f92e6d45ca911266d2f68df7f89f4da/octoprint_octoeverywhere/__init__.py#L36) as expected? Then as a result the wizard keeps popping up?

QuinnDamerell, 2020-11-10:
Not quite, what users are saying is that they can't close it on the first run.

Examples:
https://github.com/QuinnDamerell/OctoPrint-OctoEverywhere/issues/5
And
https://www.reddit.com/r/octoprint/comments/jrbwx5/new_octoprint_everywhere_bugging_octoprint/?utm_medium=android_app&utm_source=share

jneilliii, 2020-11-10:
Yeah, I just tried installing it and the wizard closed properly. I see in #5 the user doesn't have a finish button, which I have no idea would happen. I personally removed wizards from my plugins in order to avoid weird stuff that was happening and my knowledge of how it works internal to OctoPrint. You might consider doing the same thing and just move whatever logic you are doing in the wizard code somewhere else.

jneilliii, 2020-11-10:
Maybe it's a python 2.7 issue in your code? I'm running python 3 everywhere.

QuinnDamerell, 2020-11-10:
Yeah, it's so strange. The only common thread is that most users seem to be on OctoPrint 1.5, so maybe there's a bug with it or something I need to fix. 

For now I just disabled the wizard. Thanks for all of your help!

jneilliii, 2020-11-10:
OctoPrint 1.5.0 was just released today so I think this error existed prior to that, but like I said I can't reproduce the error.

cp2004, 2020-11-10:
I use two different wizards on my plugins, and have never had this issue reported. Again, couldn't reproduce with your plugin, on 1.5.0rc1. It *could* be related to the fact that no view model was bound to it, but then that would not make sense as to why it works for most.

Also, while we're talking about wizards, this part of your code:
https://github.com/QuinnDamerell/OctoPrint-OctoEverywhere/blob/d7244faf079c0471adab551394543145516d314a/octoprint_octoeverywhere/__init__.py#L32
is not required, OctoPrint handles this for you 🙂 , only showing wizards once. Shouldn't be part of the problem though.

QuinnDamerell, 2020-11-10:
@cp2004 thanks! Yeah after digging around a bit more I realized that wasn't needed. My original goal of the wizard was to actually have it show the user the link they need to click to complete the setup. But I'm such a js / py noob I couldn't figure out how to get the model to bind correctly so my code could pass the link. It's still something I want to do, but I need to find the time. Right now I'm too busy trying to fix the perf issues on the website. 👍 

Btw, do you know where the `mandatory` field is at all? I see it in the docs but I can't find it anywhere in the py or the template args. I'm wondering if explicitly setting that to false might help.

cp2004, 2020-11-10:
It's a dictionary, so would look something like this:
```python
def get_template_configs(self):
    return [
        {"type": "wizard", "mandatory": False}
    ]
```
The template vars get prefixed with `plugin_<plugin id>` to prevent name clashes. Looking at the 'OctoPrint' side of it, there are a few things that should be changed to their proper uses, but if it works..

Have you got a 'specification' for the wizard? If yes, then I could quickly try and make one for it. If it is just click the link, then verify that it works, then that should be quite easy.

cp2004, 2020-11-10:
Also, I suggest you join the [OctoPrint discord](https://discord.octoprint.org), there's a space for plugin development conversations that may be easier than a Github issue

QuinnDamerell, 2020-11-10:
Omg thanks so much! I just joined the discord, that's a great idea. :)

For the wizard, all I wanted was almost the exact same UI as the settings page. Basically just some text and a hyper link to the getingstarted page with the printer id on it. The only thing the data binding is needed for is to get the printer id to the UI. 

This is the var I was trying to get into the UI

https://github.com/QuinnDamerell/OctoPrint-OctoEverywhere/blob/81d6344bdf96662b75fe385b4842b81ec495db06/octoprint_octoeverywhere/__init__.py#L128

cp2004, 2020-11-10:
#7 Should be a good implementation, everything seems to work 🙂

jneilliii, 2020-11-10:
wizard worked for me, and for the first time I was able to connect over octoeverywhere via it's URL. This was in python 2 environment, so I wonder if I was running into issues with python 3 before?

jneilliii, 2020-11-10:
yeah, python3 environment is still not able to keep a websocket connection it seems. I'm able to register, but the printer immediately goes into "last seen x seconds ago".

cp2004, 2020-11-10:
I had this issue too. Was going to dive into it tomorrow, see if there was something that didn't look compatible (2 to 3). At least you have a wizard now 🙂

QuinnDamerell, 2020-11-11:
Yeah! I just looked into and fixed the issues with py3. I had it running on py3 a while back, but I must have forgotten to test it. 

They py3 changes are in master, let me know if you still see issues!

jneilliii, 2020-11-11:
I do see that the octoprint server is now able to stay connected in everywhere, but I can't seem to actually connect, page just never loads., I'm assuming this is server side issues, or a custom configuration issue on my part using SSL client certificates.

QuinnDamerell, 2020-11-11:
Interesting, can you grab some client logs?

jneilliii, 2020-11-11:
Sure, where can I find those?

QuinnDamerell, 2020-11-12:
I made a quick guide just for this case. 

https://octoeverywhere.com/gatherlogs

jneilliii, 2020-11-12:
Yeah, nothing of consequence it seems in there from the octoprint side.

[octoprint.log](https://github.com/QuinnDamerell/OctoPrint-OctoEverywhere/files/5528326/octoprint.log)

cp2004, 2020-12-08:
Just a heads up, we looked into this issue and there were rare cases where some settings would be returned as `None` instead of an empty string, causing an internal server error. Fixed for the next OctoPrint 1.6.0 release.

QuinnDamerell, 2020-12-20:
@cp2004 do you think that might have been the root cause here? I'm really hesitant to re-enable the wizard because I don't want users to get stuck with it open. Most people said the only way to work around it getting stuck open was to ssh into the printer and manually delete the plugin, which really sucks.

cp2004, 2020-12-20:
It was the root cause - I had the issue reported on my plugin too. However out of 3000 tracked installs, it was mentioned about 5 times - and I think it *might* have been a regression, since they were all after the 1.5.0 release.

There is a far easier way of removing the plugin, and that is through the recovery page under `http://your-ip/recovery` & restarting in safe mode.

QuinnDamerell, 2020-12-21:
Thanks for that context and thanks for fixing the bug! I'm happy I can enable the wizard again!

closed 2020-12-21 by QuinnDamerell