QuinnDamerell/OctoPrint-OctoEverywhere #9: Downloading gcode file fails - uses loopback IP instead of the octoeverywhere hostname

wmparsons, 2020-11-19:
I'll preface this with saying I'm not sure if this is actually fixable with the plugin...  I haven't looked into the source code, but I suspect the code generating the full url for the download button is outside the context of this plugin.  Reporting in the off chance it is fixable through the plugin.

Clicking on any download buttons on existing gcode files on the server fails because it uses 127.0.0.1, if I replace the IP with my octoeverywhere hostname it works fine.

cp2004, 2020-12-08:
If OctoEverywhere sets the correct forwarding headers, then this should still work as I understood it. If in doubt, you can go and ask the person who made it 🙂

QuinnDamerell, 2020-12-08:
I was just actually wondering about that! @cp2004 do you have any idea what header it would be? Perhaps the 'x-frowared-for' or 'host' headers?

I was trying to filter out as much as possible any context that would make my local calls seem different from local network OctoPrint calls, to lower the risk of some strange OctoEverywhere service header messing things up.

cp2004, 2020-12-08:
`X-forwarded-for` sounds familiar, but to properly find out get in contact with Gina & ask. (#dev-core or #dev-plugins on discord, or on the community forums) I don't actually know myself.

QuinnDamerell, 2020-12-20:
I played around with this for a while but I can't figure it out. I posted in the discord about it and we will see if anyone knows better.

I tracked down it's coming from the files API. That API is using a Flask api `url_for` to generate the hostname. I poked around for a while, but I don't see any way to tell Flask which hostname to set.

QuinnDamerell, 2020-12-21:
Fixed! 

https://github.com/QuinnDamerell/OctoPrint-OctoEverywhere/commit/edec8bc4353c562d751e919f362a4fdaebdfd074

closed 2020-12-21 by QuinnDamerell

wmparsons, 2020-12-22:
Awesome, thanks!

QuinnDamerell, 2020-12-22:
No problem! Thanks for reporting!

I did notice that some files still fail, like logs. So I will look into that. But Gcode and timelapse videos work for sure. 

(Make sure you update your plugin to the latest to get the features required)