QuinnDamerell/OctoPrint-OctoEverywhere #12: OctoEverywhere Plugin Update has locked out my Octoprint!

seipekm, 2020-12-22:
Got a notification that OctoEverywhere had an update on my Octoprint so I installed it. Now I cannot get rid of the Setup Wizard Screen on my Octoprint Web Interface. Tried recompleting the Setup and also clicking Finish. Even comes back after reboot. The Setup Wizard Screen will not go away. Please help!

QuinnDamerell, 2020-12-22:
Oh no! Sorry for the trouble. Can you send a screen shot as well as your OctoPrint version?

QuinnDamerell, 2020-12-22:
I'm sorry for your trouble, this was a past bug that I thought was resolved (see #6 ). 

@cp2004, this is so strange, do you have any ideas of how to debug this?

cp2004, 2020-12-22:
@seipekm please upload the octoprint.log, without this we cannot help debug the issue (& @QuinnDamerell I suggest creating an issue template, so you spend less time chasing logs)

I think you might have to disable the wizard. It is strange because I have not seen this very much, but with OctoEverywhere it seems to appear lots, if it is the same issue (we won't know until we get the logs 😉).

The problem is mitigated in OctoPrint 1.6.0, but obviously this has not been released yet so the fix is not there.

@seipekm to 'fix' the install, you can head to the recovery page `http://<your-ip>/recovery` and restart in safe mode.

QuinnDamerell, 2020-12-22:
Thanks, @cp2004. I have been following up on our reddit [here ](https://www.reddit.com/r/OctoEverywhere/comments/khxshr/octoeverywhere_plugin_update_has_locked_out_my/) as well, hopefully, someone will find the time to grab some logs for us. 

I wish I could repo it, but I can't. I wonder if it's much more likely to happen on some older version of OctoPrint or something.

cp2004, 2020-12-22:
We were never able to reproduce either - the root cause was with gcode scripts, because the settings are saved when gcode scripts were saved.

That reddit post has far more comments than I expected it too! Damn, really want to know what causes it in the first place.

We *only* saw it after the release of 1.5.0. Even though the code that the problem came from was not touched in between updates (& so therefore was not deemed regression - would have been good to get fixed in a patch/release candidate) something seems to trigger this more now.

QuinnDamerell, 2020-12-22:
Yeah, it's really bugging me as well. It must be something somewhat unique to my plugin because it does seem to repo super often compared to others.

cp2004, 2021-02-01:
Apparently, it is the PSU control plugin https://github.com/kantlivelong/OctoPrint-PSUControl/issues/162

All this time, and I have only just found the issue.

QuinnDamerell, 2021-02-01:
@seipekm - by any chance, do you remember if you had the PSUControl plugin installed when you had the issue?

QuinnDamerell, 2021-02-02:
It sounds like from the linked threads that this has been resolved by the plug that seemed to be causing the issue. I will wait for a little while for the other plugin to rollout, and then re-enable the wizard in this one. :)

closed 2021-02-02 by QuinnDamerell