QuinnDamerell/OctoPrint-OctoEverywhere #13: Octoeverywhere crashes octoprint after a few seconds of connecting

berkeoznalbant, 2021-01-10:
As I have said in the title. Octoprint crashes each time I connect it with octoeverywhere. [Here](https://github.com/QuinnDamerell/OctoPrint-OctoEverywhere/files/5793226/octoprint.2.log) is the log. On line 3595 it crashes in the logs.

QuinnDamerell, 2021-01-10:
Thanks for the report! That's very strange, I don't think the plugin should be able to crash OctoPrint at all. What hardware are you running on?

berkeoznalbant, 2021-01-10:
I'm running raspberry pi 3B+ with original adapter. After octoprint crashing, I can restart it with SSH connection from laptop.

QuinnDamerell, 2021-03-14:
Closing this for now due to its age. Try installing the latest version and let me know if you are still having trouble!

closed 2021-03-14 by QuinnDamerell