QuinnDamerell/OctoPrint-OctoEverywhere #14: Setup not working

mmaslovpy9i, 2021-02-19:
I clicked the "finish setup" button and it brought me to this page:

https://octoeverywhere.com/getstarted?printerid=4TZVDFCTPHW4L8ZNF38LOAIKZW9Z0FOMQQXTS23U

Then I've got "printer id not found" error in the screenshot attached.

[image]

Please help!

QuinnDamerell, 2021-02-19:
Hey @mmaslovpy9i! Can you follow the steps here https://octoeverywhere.com/gatherlogs and send me an email at support@octoeverywhere.com so we can chat in private? Thanks!

QuinnDamerell, 2021-02-23:
Hey! Let me know if I can help you out! If not, I will close this for now. :)

closed 2021-02-23 by QuinnDamerell