QuinnDamerell/OctoPrint-OctoEverywhere #15: updated to latest plugin no 2nd cam with multi cam

swheettaos, 2021-04-12:
Updated to the latest version.  Multicam still not working :( 

first cam is /webcam/?action=stream
second cam is /webcam2/?action=stream 

in multicam setup.  Works fine when on LAN .. but no workie when over internet.

Here is system info:
browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/89.0.4389.114 Safari/537.36
connectivity.connection_check: 1.1.1.1:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 3959291904
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.17.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 21.0
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.6.0rc1
printer.firmware: Marlin bugfix-2.0.x (Mar 26 2021 05:52:09)
systeminfo.generator: systemapi

here is system bundle 
[octoprint-systeminfo-20210412104947.zip](https://github.com/QuinnDamerell/OctoPrint-OctoEverywhere/files/6299776/octoprint-systeminfo-20210412104947.zip)

QuinnDamerell, 2021-04-12:
Thanks for the feedback! Right now that's the only config that might now work, because it uses relative URLs. But I'm thinking about a way I can fix it, I might try to roll it out soon.

I would assume the first camera is working but the second isn't, is that correct?

If you want to work around it, you can use the IP of the printer to make an absolute URL. So instead of using `/webcam/?action=stream` use something like `http://192.168.1.50/webcam/?action=stream` where `192.168.1.50` is your device IP.

QuinnDamerell, 2021-04-16:
Hey! The latest plugin update `0.8.2` should fix this issue! Please give it a go and let me know if you have any issues!

closed 2021-04-16 by QuinnDamerell