QuinnDamerell/OctoPrint-OctoEverywhere #17: Webcam feed not working after update

Onejk1, 2021-04-21:
After updating to latest version webcam will not work via octoeverywhere. It is working fine on octoprint locally just not via anything using octoeverywhere.

was working perfectly fine before the update stream location is
/webcam/?action=stream

QuinnDamerell, 2021-04-21:
Sorry to hear you're having trouble @Onejk1! Let's fix it. 

Can you gather OctoPrint logs using this guide:
https://octoeverywhere.com/gatherlogs

And then open a support ticket here:
https://octoeverywhere.com/support

Thanks!

mytja, 2021-06-09:
I also experience this issue :disappointed:

QuinnDamerell, 2021-06-09:
Thanks for reporting @mytja! Can you also open a ticket and send logs?

Gather OctoPrint logs using this guide:
https://octoeverywhere.com/gatherlogs

And then open a support ticket here:
https://octoeverywhere.com/support

I'm happy to help!

mytja, 2021-06-18:
I found the culprit for my issue
I was using old url - https://myprintername.octoeverywhere.com
Meanwhile i found new url - https://myprintername.lon.octoeverywhere.com

QuinnDamerell, 2021-06-19:
Interesting, either should work for you. What happens when you use the old URL? What response do you get back?

mytja, 2021-06-19:
> Interesting, either should work for you. What happens when you use the old URL? What response do you get back?

I got a MJPEG test page, I showed you in my first comment.

I also noticed, that ALL POST requests didn't work for me anymore. So I couldn't set the temperature, start printing, uploading files etc.

QuinnDamerell, 2021-06-23:
Thanks for the info! Can you make sure your plugin is updated to the latest version? If you still have problems, can you send me your OctoPrint logs using this?

Gather OctoPrint logs using this guide:
https://octoeverywhere.com/gatherlogs

And then open a support ticket here:
https://octoeverywhere.com/support