QuinnDamerell/OctoPrint-OctoEverywhere #19: Annoying Snapshot Errors in OctoPrint log

unixneo, 2021-09-29:
I noticed that this plugin keeps trying to make web cam snapshots and there is no way to disable this in the plugin, so I was forced to comment out that code to stop these errors in the log.

See:

https://community.unix.com/t/day-30-adventures-in-3d-printing-for-beginners-manual-print-bed-leveling-v-software-mesh-leveling/384047/5

QuinnDamerell, 2021-09-29:
Thanks for the feedback! I will admit I'm quite a noob to writing python, so I appreciate any feedback or help I can get!

Can you clarify for me, which error were you seeing logged? Was it 

```NotificationsHandler failed to send event```

Or 

```Snapshot http call failed```

Thanks!

unixneo, 2021-09-30:
Hi Quinn.   Hope all is well.  

In my mind, the overarching issue is not which error message generates, it is a software design issue.  The issue is not related to Python or any programming language, because the issue is a design (logic) choice.

If I may paraphrase, your plugin code (related to snapshots) works something like this:

> Periodically try to send a snapshot (regardless of if the user wants a snapshot or not, or has a web cam or not) and if no web cam is enabled or present, log an error.

I think your design should be:

> Check a user plugin setting to see if snapshots are enabled.  If enabled then "try to send a snapshot and if no web cam is enabled or present (or some other issue), log an error."  If not enabled, completely skip trying to send a snapshot.

This is what I have done by commenting out this code in your plugin.  I commented out this code in your plugin because I do not have a web cam attached and I do not want OctoPrint  (via your plugin) attempting to send a snapshot (continuously) and logging errors because no web cam is present.

If you simply add a setting for your plugin for a boolean (checkbox), something like:

```
Enable Snapshot  (yes/no)
```

and you wrap the code I have commented out with this boolean (a simple if statement), the issue is resolved.  The user can select to enable or disable snapshots.

In summary, this issue is not related to the programming language nor the exact error returned when the plugin attempts to take a snapshot.   The issues exists because of a design choice to not provide a plugin setting where the user can turn snapshots on or off.   I think it (my opinion) this was not the correct "software design choice" to not include a plugin setting which permits the user to enable or disable a snapshot.  Instead, the design / logic of this plugin is to just attempt to access the web cam (continually) and attempt to send snapshot continually and log an error if there is no web cam.

Better, in my mind, is to add a boolean plugin setting and default this to "false" and then users can enable the snapshot feature if they wish.

Hope this helps.

unixneo, 2021-09-30:
On another issue, while we are on the topic of OctoEverywhere!  user settings.

[image]

```shell
2021-09-30 14:50:31,381 - octoprint.plugins.octoeverywhere - INFO - Attempting to talk to OctoEverywhere, server con wss://starport-v1.octoeverywhere.com/octoclientws[xx]
2021-09-30 14:50:31,601 - octoprint.plugins.octoeverywhere - INFO - Connected To OctoEverywhere, server con wss://starport-v1.octoeverywhere.com/octoclientws[32]. Starting handshake...
```
It would be good for this plugin to recognize that we are already synced / logged in and not display a 

```
Finish Your Two Minute Setup Now!  
```

After we are fully set up and using both OctoEverywhere! and OctoPod daily.

It's good to pay attention IMHO, to the plugin user settings, add the setting I mentioned earlier and correct the status of the user.

QuinnDamerell, 2021-09-30:
Yes! Thanks for that feedback! 

Your right about the eventing system. When I was creating it I was debating between the system I have now where it always pushes all events and a system where the plugin would get the desired user event config from the service and only send when the user asks. Ultimately for my v1 I decided to go with the first one because it's not nearly as efficient, but it's easier to implement and given all of the work I had to add on the server-side, I wanted to simplify things. However, given some of the user feedback I have gotten, I do think the "only do work when configured to do so" approach would be better, so I will probably move to that soon.

You're also right about the webcam. For the first version, again for simplicity, I made the logic always try to get the image. But it would be more ideal to allow the user to enable/disable it.

Last, about the settings UI, your right again! I have had fixing that on my backlog for a long while and it really bugs me. But for most users, they never seem to go back to that settings page once set up, so it doesn't really matter if it updates or not. But again, it's something I would really like to fix, when I find the time.

Overall, thanks for your feedback! In the short term, I will disable at least printing that error if there's no webcam setup or it's not available. Longer-term, I hope to make the system better by adding the features I described above.

closed 2021-09-30 by QuinnDamerell