QuinnDamerell/OctoPrint-OctoEverywhere #20: Plugin not downloading

d-portero, 2021-10-08:
Trying to download the plugin from the plugin manager. Does not appear when searching for "octoeverywhere" and installation through the URL says the download completes. After restart, the plugin is not visible in the list of downloaded plugins

QuinnDamerell, 2021-10-08:
Thanks for reporting! Can you send me logs? Here’s how:
https://octoeverywhere.com/gatherlogs

d-portero, 2021-10-08:
There is not an option to collect logs from octoeverywhere. Seems it is not being downloaded at all

QuinnDamerell, 2021-10-09:
Thanks for the logs! They don’t seem to capture the problem, sometimes the OctoPrint logs do that. Can you reproduce the issue again, restart OctoPrint, and then capture them again? Sorry for the trouble.

d-portero, 2021-10-10:
Here you go. I downloaded it, restarted, downloaded it, and then restarted. 

[octoprint (1).log](https://github.com/QuinnDamerell/OctoPrint-OctoEverywhere/files/7318514/octoprint.1.log)