QuinnDamerell/OctoPrint-OctoEverywhere #23: Either remove promotional OctoPrint UI notifications or provide a way to permanently disable them

bcat, 2021-12-16:
First of all, I want to say I quite like OctoEverywhere overall and am totally fine with the paid aspects as well as the need to support the service more generally, so please don't take this as too harsh a criticism. That said, the notifications the plugin gives for the "birthday celebration" are honestly not ideal, in my opinion. Notifications in the OctoPrint UI should either tell the user things about ongoing prints, safety issues, or updates. Using them for promotional things (even free promos) really leaves a bad feeling for me. Imagine if every OctoPrint plugin did this; it'd be impossible to use the UI.

If it was just one notification, I wouldn't have filed a bug, but I just received what I believe is the third one I've seen in the OctoPrint UI, and that's enough for me to file this. My personal suggestion would be just removing these notifications entirely, as they don't seem to be an appropriate use of the OctoPrint notification system. But if you end up keeping them, can you please add an option in the plugin settings to permanently disable them on a per-user basis? Maybe not everyone feels strongly about this, but I also suspect I'm not the only one who wants to disable them.

QuinnDamerell, 2021-12-16:
I'm sorry, that wasn't intended. I just released a server update that will stop this from happening. The problem was due to a quick fix I put out earlier today to mitigate an issue with a 3rd party service I use. 

The 3rd party service issue and the portal notification issue have been resolved, I'm sorry for the trouble it caused. I only built in the portal notification system for very rare events, since I 100% agree that getting notifications often is really annoying.

closed 2021-12-16 by QuinnDamerell

bcat, 2021-12-16:
Thanks for the quick response! Glad to hear it wasn't an intended effect, and thanks for understanding. :)