QuinnDamerell/OctoPrint-OctoEverywhere #25: OctoEverywhere plugin, after restart octoprint server or reboot, causing safe mode state on my raspberry pi

hansvanlin, 2022-06-22:
See issue desciped at https://www.facebook.com/groups/piprint/posts/1445501955888523?comment_id=1446021195836599&reply_comment_id=1446022305836488&notif_id=1655837808278970&notif_t=group_comment&ref=notif

I see that more people have the same issue

hansvanlin, 2022-06-22:
You want to live debug it or should I provide you with logging or configuration?

QuinnDamerell, 2022-06-23:
I debugged this all night and I think I know the cause. The issue is something in my plugin and also a problem with how OctoPrint handles settings. I fixed up the logic in my plugin to mitigate the issue and I think it should be super rare to have this happen now to users. I also opened an OctoPrint bug report to track a possible actual fix, which would have to be in OctoPrint and it would benefit all plugins!

OctoPrint Bug:
https://github.com/OctoPrint/OctoPrint/issues/4551

I will release the fix in a few days, but it's in the repo now. If you want to try it ASAP, you can follow this guide:
https://octoeverywhere.freshdesk.com/support/solutions/articles/69000790524-manually-updating-the-octoeverywhere-plugin

Let me know if the fix works for you!

closed 2022-06-23 by QuinnDamerell

hansvanlin, 2022-06-23:
I found a spare 30min to test it. Reinstalled the plugin and did serveral octoprint server restarts and system reboots and I could not reproduce the safemode issue.
Thanks for the update!
Best regards,
Hans

QuinnDamerell, 2022-06-23:
Amazing! I'm so glad that worked for you! Please let me know if you hit the issue any more in the future!

hansvanlin, 2022-10-11:
Thanks, I will do 😀