QuinnDamerell/OctoPrint-OctoEverywhere #26: OctoEverywhere 1.9.4 impacts timelapse quality with OctoLapse

mathieucarbou, 2022-07-17:
I am using OctoLapse and OctoEverywhere.

Before 1.9.4, everything was working fine. OctoLapse was responsible for the  camera settings (in my case 480p 10 fps) and most importantly responsible to set a hight enough variable bitrate to not have any artifacts. 
And OctoEverywhere was then streaming fine.

Since 1.9.4, by image quality is completely annihilated because OctoEverywhere seems to force a very low bitrate for the video.

As an OctoEverywhere subscriber, I must say I am not very happy about this change. Even if there is a disclaimer from OctoLapse telling that OctoEverywhere does not play nice with OctoLapse, there is a limit... ;-) If OctoEverywhere wants to take responsibility of the camera settings (which is not his role IMO), then we should have an option to completely disable any camera streaming from OctoEverywhere and only use the other features but not te camera stream.

This can be seen with all the color artifacts in the image:

[image]

Could you please explain how to revert to the previous version please until this is fixed ?

mathieucarbou, 2022-07-17:
Just to add some details: I started OctoPrint today on my RPI, it asked for OctoEveywhere update, which I did, restarted then  the camera issue came.
I've tried in OctoLapse to play with some settings but it's like OctoLapase does not have anymore any control over the stream: changing the brightness / bitrate / exposure time / etc does not affect the stream anymore.

QuinnDamerell, 2022-07-17:
Thanks for bringing this up! I think I know the cause here, the fix should be easy. I this should only be effecting the dash board view, actual timelapse should be fine. Can you do a quick test print and see if that's the case?

Can you also tell me what your camera setup is like? What camera are you using and at what resolution?

mathieucarbou, 2022-07-17:
I am using the raspi cam v2 with the `bcm2835-v4l2` module and the following config:

```
camera="usb"
camera_usb_options="-x 640 -y 480 -fps 10 -quality 100"
camera_http_webroot="./www"
camera_http_options=""
```

I've followed the OctoLapase user guide, but set a lower frame rate and resolution because I don't need more and also want to have a faster stream with OctoEverywhere.

The thing is that with the raspicam, the bitrate is really important to control because the jpeg quality or some other settings do not work. OctoLapse guide explains this very well. We have to increase to a variable bitrate to ~ 6-8 M to have something good.

mathieucarbou, 2022-07-17:
By disabling both plugins, reactivating first OE, then OL, resetting the camera settings and putting them back again, this works as before now.

[image]

mathieucarbou, 2022-07-17:
I also set back to: `camera_usb_options="-r 640x480 -f 10"` (removed quality which was just a temporary try - I was running before with these settings seen here)

QuinnDamerell, 2022-07-17:
Very strange, can you send me a log file? Don't send it here, as it might have personal info in it.

https://oe.ink/s/logs

Send it to: support@octoeverywhere.com

mathieucarbou, 2022-07-17:
I've started ~4h print. As soon as I can finish it I will :-)

QuinnDamerell, 2022-07-17:
Sounds great! No problem at all. I will keep digging into what could have happened on my side as you do.

Do you have OE notifiaitons enabled? Can you let me know if any of them look incorrect?

mathieucarbou, 2022-07-17:
Note that in the case of OctoLapse, is is fully taking control over the camera settings, including those of the dash board view. When I configure the webcam settings through OctoLapse it also impacts the stream.

mathieucarbou, 2022-07-17:
I have the OE notifications enabled, yes, but just emails.

I have started a 4h live link here (https://octoeverywhere.com/live/HEDZaztvll), with the settings above and raspi cam. Works pretty good and fast. 

But for this print I've disabled OL because the print quality was impacted too much when using the stabilising option required for the location of my cam.

mathieucarbou, 2022-07-17:
I think we could close... I did several prints with timelapses and shared links today and everything was working fine. 
I am completely clueless about what exactly I did that could solve that though...

QuinnDamerell, 2022-07-21:
No problem! Thanks for all of the debugging, please feel free to re-open this if there's any issue in the future!

closed 2022-07-21 by QuinnDamerell