QuinnDamerell/OctoPrint-OctoEverywhere #30: Running out of memory during printing with octoeverywhere enabled

hansvanlin, 2023-01-21:
During a print my rpi runs out of memory because  slipstream is cached untill memory is full. The octoprint freezes and serial connection is lost and print stops suddenly.
When plugin is disabled the serial connection is good and the print doesn´t stop.
added a screenshot of the log and memory just before it is full and a movie where the plugin is disabled where memory value is constant.

octoprint 1.8.6
octoeverywhere 1.11.1

[image]
[image]

https://user-images.githubusercontent.com/108009649/213881386-d64a3df6-125a-48f7-91d4-5cfbd8304b35.mp4

QuinnDamerell, 2023-01-25:
Hey! Thanks for opening this issue. Can you please submit a support ticket to discuss this more privately? support@octoeverywhere.com

closed 2023-01-25 by QuinnDamerell

hansvanlin, 2023-03-10:
Hi Quinn, this week I tested and installed octoeverywhere 2.3.7 and octoprint 1.8.7 and found that this issue is still there.
I started a print and looked via ssh at the memory usage and it increased untill it crashed.