QuinnDamerell/OctoPrint-OctoEverywhere #35: OctoEverywhere - First Layer Complete Notification Improvments

QuinnDamerell, 2023-02-22:
Hello All!

Today I'm excited to announce major fixes and updates to the "first layer complete" notification. This notification has been problematic in the past due to the plugin not knowing the exact layer height. While it still doesn't know the exact layer, the new system I put in place should be much more accurate, and thus it will fire the notification at the correct time!

While I was at it, I added a "third layer complete" since I have heard that some makers would rather be informed after the first few layers to take a look rather than just one. It's up to you which you get, one layer, three layers, or both!

Both of these notifications can be controlled for any platform in the notification settings, found here:
https://octoeverywhere.com/notifications

Happy Printing!

closed 2023-02-27 by QuinnDamerell

ForceConstant, 2023-05-09:
Just tried this out, and I did get a premature first layer notification, I am guessing due to some jogs before I start printing. What does OE use to determine when we are actually starting printing? I am on Mainsail.

QuinnDamerell, 2023-05-15:
@ForceConstant it's based on observation of the print, but it needs to have exact knowledge to know when the first layer is done. Can you tell me what your gcode setup is like? Do you have any special startup logic?