QuinnDamerell/OctoPrint-OctoEverywhere #36: Multiple camera confusion

ShapeShifter42, 2023-02-22:
This sounds similar to an old issue but I think it's the opposite problem: [Issue #10 - Multiple Cameras](https://github.com/QuinnDamerell/OctoPrint-OctoEverywhere/issues/10)

In my Octoprint setup, I have two web cameras. I would like to use the main camera (camera 1) with OctoEverywhere. But it seems to be using both cameras, at different times:

- When I check in with the quick view on the OctoEverywhere website, it shows the primary camera, which is what I want
- When I look at the OctoEverywhere dashboard, the thumbnail with the the quick view button shows the second camera, which is not what I want
- When I get a print job complete notification, it has a snapshot from the second camera
- When looking at quick view, seeing the primary camera, if I click the button to report a missed print failure, it shows me a picture from the second camera and asks what is it seeing? While the primary camera shows the print failure, the second camera doesn't show a view that shows the print problem, so there is nothing bad in that picture.
- I've had a pretty bad print failure, and while looking at the quick view, the spaghetti is clearly visible. But gadget thought everything was fine, presumably because it was watching the second camera that can't see the problem.

I don't see any kind of setting to select which camera should be used. I'd like OctoEverywhere to _only_ use the primary camera for everything (especially Gadget) but I see no such option. Barring such an option, I'd assume OctoEverywhere would use the primary camera, but it seems to be randomly selecting a camera for its different purposes.

OctoEverywhereDocs, 2023-02-23:
@ShapeShifter42, thanks for opening this issue! I was looking at a similar issue, and found that the user had setup the webcam streaming URL, but didn't setup the webcam snapshot URL for each instance. Can you go into each instance and check that the Webcam snapshot URL is also set correctly?

ShapeShifter42, 2023-02-24:
Thank you for the fast response. I'm far from an OctoPrint expert, but I'm leaning fast. I'm not sure I properly understand your suggestion.

I set up the second camera using this page, which I found to be very helpful: https://community.octoprint.org/t/setting-up-multiple-webcams-in-octopi-the-right-way/32669 

Looking at the OctoPrint configuration pages, under Webcam and Timelapse, I see that the timelapse snapshot URL was http://127.0.01:8081/?action=snapshot - I don't remember changing it, but clearly I must've done so, as that URL yields a snapshot of the secondary cxamera. I've changed it back to http://127.0.0.1:webcam/?action=snapshot, which yields a snapshot of my primary camera - the one I want to use with OctoEverywhere.

Assuming that Gadget is periodically grabbing a snapshot to do its analysis, which would make sense, then I'm sure it will help the problem. That could also explain why sometimes it shows the secondary camera (in those cases where it's showing a still picture) and sometimes the primary camera (when it's showing a video.) 

This makes sense, but I can't test it at the moment as it's getting late and I've already shut down the printer for the night. I have more printing to do tomorrow, so I will check it out then. 

Is this what you meant by checking the snapshot URL? Perhaps, although I'm not sure what you mean by "each instance" as I only see the one entry field in the OctoPrint settings. If I should've done something else, could you please be a little more specific where these settings may be? Like I said, I'm not expert in OctoPrint, but I'm learning _fast_.

ShapeShifter42, 2023-02-24:
I guess I was tired last night - thinking about it with a clear head this morning, I do remember setting the snapshot URL to the second camera. I have OctoLapse set up to take a snapshot on the front facing main camera on every layer change, and I set the regular timelapse snapshot URL to the top mounted down-facing secondary camera to make a timelapse based on time. This gives me two different views: one in front showing the model building smoothly, and one from above showing all of the movement of the printhead. The time based downward view is good for wide thin prints that take a long time, but don't have a lot of layers, causing OctoLapse to generate a very short timelapse.

So, it looks like OctoEverywhere is making use of the default timelapse snapshot URL for Gadget? If so, it looks like I can't have it both ways - having Gadget work properly is more important to me, so given the choice, I will give up that timelapse on the secondary camera. (Maybe I can manually set up an independent script to take snapshots and build a timelapse.)

Overall, I'm very impressed with OctoEverywhere, and I'm happy to be a paid supporter. 👍But I do have one minor pet peeve about the UI: there is a banner at the top of the page that says that Gadget is free and I should try it now - I do have it enabled, and I wish that banner would go away once it's enabled. It makes me think that something got changed and it's no longer enabled. 😉

QuinnDamerell, 2023-02-27:
Sorry for the delay; that's excellent details! You're right; OctoEveywhere uses the webcam stream URL and the snapshot URL. Gadget uses the snapshot URL because it's a much more efficient way to get the camera view over the webcam stream. I do need to figure out a better way to inform users about setting the snapshot URL, a lot of users don't update it correctly. 

I love to hear that you're enjoying the service! Your feedback about the Gadget banner is great! I will definitely look into fixing that.

Please let me know if there's anything else you find you don't like about the service, I love to hear even the little things!

closed 2023-02-27 by QuinnDamerell