QuinnDamerell/OctoPrint-OctoEverywhere #37: OctoEverywhere - Multi Printer Support!

QuinnDamerell, 2023-03-18:
Big news... with version 2.4.0, our plug-in now supports remote access for multiple Klipper/Moonraker instances running on the same device!

We are on a final push toward our big launch, so pound on the plug-in hard and be sure to let us know if you have any issues or feedback! 🚀

pauluchin, 2023-03-18:
Thanks 4 the good job and support for klipper

Higachikunideguchi, 2023-03-18:
After that update i cant connect to mainsail localy, even reinstall all linux not help... wtf>

QuinnDamerell, 2023-03-19:
@Higachikunideguchi - thanks for letting me know! Are you able to ssh into the device?

Higachikunideguchi, 2023-03-20:
I repaired it on Saturday - i dont know why and when, but something close
porta for communication. I use Linux Mint on terminalPC instead of
RaspberryPi, do its little harder to solve. Propably the problem was on
system - after installing New system same problem, fixed manualny.

closed 2023-03-27 by QuinnDamerell