QuinnDamerell/OctoPrint-OctoEverywhere #38: Broken web view in OrcaSlicer

TexZeTech, 2023-04-03:
As of this time with OctoEverywhere  2.4.0-4-gaca09d5  in Klipper Mainsail

It seems the device tab in [OrcaSlicer 1.5.0 Appimage Linux](https://github.com/SoftFever/OrcaSlicer) has been broken after installing the addon OctoEverywhere to my printer. 

I have yet to remove OctoEverywhere to try and confirm this suspicion as I was wondering if anyone else could confirm before I have to make any changes to my system.

QuinnDamerell, 2023-04-04:
Thanks for letting me know! I'm not familiar with OrcaSlicer. Is that something you run on a computer in the same network or on the same device?

TexZeTech, 2023-04-04:
It's the slicer I use. It runs on the computer not on the printer but it has a tab that allows remote management like a web browser.

I still haven't ruled out if it's something on my end which is why I wanted to know if anyone else was able to reproduce.

closed 2023-04-04 by TexZeTech