QuinnDamerell/OctoPrint-OctoEverywhere #40: Install script targets wrong moonraker.conf and directory in RatOS.

miklschmidt, 2023-04-13:
Hi! I have several users reporting problems with the octoeverywhere install script modifying the RatOS git repository.
Instead of finding `/home/pi/printer_data/config/moonraker.conf` it finds `/home/pi/printer_data/config/RatOS/moonraker.conf` (which is included by /home/pi/printer_data/config/moonraker.conf) modifies that, and dumps it's files in `/home/printer_data/config/RatOS/`

Any particular reason it's doing a depth first search here? Or is it looking for specific content in moonraker.conf? What's going on?

miklschmidt, 2023-04-17:
@QuinnDamerell Awesome! let me know when i can release the horde for a test run! :D

miklschmidt, 2023-04-17:
Also if there's anything else i can help with, please let me know :)

QuinnDamerell, 2023-04-17:
@miklschmidt no problem! Can I ask, do you know of anyone who runs multiple instances of moonrakeer and klipper on one RatOS device?

miklschmidt, 2023-04-17:
@QuinnDamerell None, it's currently not supported :)

QuinnDamerell, 2023-04-17:
@miklschmidt perfect! Ok, I pushed the new installer to master! I explicitly tested it on RatOS to ensure it discovered things correctly, and it looks good! 

If you find a minute, pull the current repo and run the ./install.sh script again to see if it works for you! If it's currently installed in the wrong location, it should be re-install into the correct place.

miklschmidt, 2023-04-17:
@QuinnDamerell i ran it on my test machine, and it looks good to me! I've asked the users to give it a go again, i'll let you know if i hear of anymore problems but for now i consider it fixed :)

closed 2023-04-17 by miklschmidt

QuinnDamerell, 2023-04-17:
Perfect! Thanks for bringing this up and please let me know if you hit any other issues in the future! 

Btw, the installer is safe to run anytime, even for recovery and such. But in this one case, users might have to re-link their printers, because a of files moving around.

miklschmidt, 2023-04-17:
Excellent, i've issued a PSA on discord letting people know, thanks for the fix!