QuinnDamerell/OctoPrint-OctoEverywhere #41: Feature Request: Divide out notifications between errors

mikebcbc, 2023-05-11:
Hi there,

I would only like to receive notifications for Gadget detections/errors - not all errors. Is there anyway for webhooks and desktop that notifications can be split up to "Gadget Error" and "Print Error"? The reason for this is that I use emergency stop quite often and I don't want to receive the notification for that error all the time - only if Gadget finds an error.

QuinnDamerell, 2023-05-15:
@mikebcbc, that's excellent feedback. Right now, you could disable all of the error types in the notification settings, and then you would only get Gadget errors. Right now, there's no way to turn off the Gadget notifications if Gadget is enabled to run. But I will add features to fix that to my TODO list!

closed 2023-05-15 by QuinnDamerell

mikebcbc, 2023-05-15:
> @mikebcbc, that's excellent feedback. Right now, you could disable all of the error types in the notification settings, and then you would only get Gadget errors. Right now, there's no way to turn [...]

@QuinnDamerell - Oh, that's good to know!! Just to confirm then - if I uncheck all boxes in notification settings - including "Print Failed" - but I have Gadget enabled, I would still be receiving notifications for Gadget? Thanks Quinn!

QuinnDamerell, 2023-05-15:
@mikebcbc yup! That's correct! The UI doesn't indicate that very well, but as long as the enabled switch is set to true for the endpoint, Gadget will send notifications. It's independent of the print failed notification. I need to improve the UI to make it more clear. :)

mikebcbc, 2023-05-15:
> @mikebcbc yup! That's correct! The UI doesn't indicate that very well, but as long as the enabled switch is set to true for the endpoint, Gadget will send notifications. It's independent of the print failed notification. I need to improve the UI to make it more clear. :)

You rock! Thanks Quinn.