QuinnDamerell/OctoPrint-OctoEverywhere #42: Klipper/Mainsail - Webcam unable to load with Crowsnest v4 camera-streamer & webrtc

Edigorin, 2023-05-24:
After changing to camera-streamer & webrtc option in Crowsnest v4, the webcam is unable to load in Octoeverwhere. Webcam work fine in Mainsail itself

QuinnDamerell, 2023-05-24:
@Edigorin, thanks for letting me know! I haven't tried that setup yet. Is there a guide or something you followed to setup v4?

Edigorin, 2023-05-24:
Here's the short explanation from the Mainsail Discord, in the announcement channel:

A short guide to updating Crowsnest to v4:
```bash
cd ~/crowsnest
make uninstall

# update Crowsnest via update manager

cd ~/crowsnest
sudo make install
```
To use the new streamer, you have to change the mode in `crowsnest.conf` from `mjepg`/`ustreamer` to `camera-streamer`. After Crowsnest restart, you have to change the mode in Mainsail webcam settings to `WebRTC (camera-streamer)` and change the streaming URL to `/webcam/webrtc`.

Here's the link to the post: https://discord.com/channels/758059413700345988/781284268654854154/1110660341693677650

Edigorin, 2023-05-31:
Here's an updated guide on installing V4:
https://crowsnest.mainsail.xyz/faq/upgrade-from-v3-to-v4

QuinnDamerell, 2023-05-31:
Thanks for the update! I just pushed an update to the plugin yesterday that fixes some of the issues, but the WebRTC stream in mainsail still won't work. I think for now, I might have to force mainsail to use the old streaming stack only when connecting via OctoEverywhere. I know what I need to do in-order to make WebRTC work, but it requires changes in the mainsail logic, which I'm just starting the conversations for now.

QuinnDamerell, 2023-06-12:
I'm still working with the `camera-streamer` dev on a full solution here, hopefully it will be ready soon!

QuinnDamerell, 2023-07-14:
Update, this is still on-going. I'm still waiting on the deployment of the new packages of `camera-streamer`.

flow23, 2023-12-23:
Hi, is this still worked on?

QuinnDamerell, 2023-12-26:
Hey! Yeah, I need to follow up with the crowsnest and camera streamer people. The OctoEverywhere logic is already done and rolled out, so whenever it's completed on their end, it should all work!

arcaspo, 2024-01-28:
Hi, i am still encountering this issue but only on mobile, when i view the webcam through octoeverywhere on desktop it shows up fine but on mobile the stream doesn't work.

QuinnDamerell, 2024-10-24:
I know it's been a while, but this is still in progress. The upgrade to the latest camera streamer seems to have stalled due to other projects out of my control, so I'm waiting with the rest of you. :)

prettyyellowdog, 2025-04-27:
How's the progress on this?  It would also help to get the camera functionality up and going on the Creality K2, which also uses webrtc.

QuinnDamerell, 2025-05-15:
I'm still working on this, but the K2 and Creaility Hi are different WebRTC setups, so they are their own monsters. :D

NegussugeN, 2025-06-25:
Are there updates to this? I am experience the same issues as the OP, both with the camera-streamer/ WebRTC feed not loading either in Mainsail or within OctoApp when they are accessed remotely.

closed 2026-09-17 by QuinnDamerell