QuinnDamerell/OctoPrint-OctoEverywhere #43: _DoAutoSettingsUpdate trying to add a "/" before "stream_url" value in "octoeverywhere.conf"

wattsie, 2023-06-10:
Hi all,
Got a weird issue, when running octoeverywhere, I find the CAM disappears.

In the config file I set:
`snapshot_url = http://10.0.0.1/webcam/webrtc`

But when is launches, the URL keeps getting reset to:
`snapshot_url = /http://10.0.0.1/webcam/?action=snapshot`

I have set the file to read-only, so it is working at the moment, but every few minutes the log entry is shown saying permission denied (obviously we the conf file is read-only atm) 

If I change the perms back to writable, the conf file changes back with the leading "/" and the webcam stops appearing on octoeverywhere dashboard.

I then go back in, set file to read-only, force update the URL to remove the leading "/" and HUP the process, and tada the webcam shows again on the dashboard.

```
2023-06-10 14:08:23,205 - INFO - Webcam helper found a snapshot url of [/http://10.0.0.1//webcam/?action=snapshot] but detected WebRTC for the stream URL, so we are updating the URL to the camera-streamer snapshot URL
2023-06-10 14:08:23,226 - INFO - Webcam helper updated webcam settings. streamUrl: /http://10.0.0.1/webcam/stream, snapshotUrl: /http://10.0.0.1/webcam/snapshot, flipH: False, flipY: False, rotation: 0
2023-06-10 14:08:59,586 - INFO - Main request http URL threw an exception: No connection adapters were found for '/http://10.0.0.1/webcam/stream'
2023-06-10 14:08:59,587 - ERROR - Main request failed and we have no more fallbacks. We DON'T have a main response.
2023-06-10 14:08:59,590 - WARNING - Web Stream http [86]  failed to make http request. octoHttpResult was None; url:/webcam/?action=snapshot
2023-06-10 14:09:04,530 - INFO - Main request http URL threw an exception: No connection adapters were found for '/http://10.0.0.1//webcam/stream'
2023-06-10 14:09:04,531 - ERROR - Main request failed and we have no more fallbacks. We DON'T have a main response.
2023-06-10 14:09:04,533 - WARNING - Web Stream http [88]  failed to make http request. octoHttpResult was None; url:/webcam/?action=snapshot
```

What am I doing wrong?

Many thanks in advance.

QuinnDamerell, 2023-06-12:
Thanks for the bug report! I'm working on full WebRTC support, but as a workaround right now, the system is setup to fallback to the stream API. Why it's adding a leading slash I have no idea, I will look into that. 

Can you open a ticket at https://octoeverywhere.com/support so we can chat more privately with private info?

QuinnDamerell, 2023-07-14:
@wattsie and others helped me debug and fix this; thanks!

closed 2023-07-14 by QuinnDamerell