QuinnDamerell/OctoPrint-OctoEverywhere #45: Missing certain notifications

pifiuk, 2023-07-11:
Prusaslicer 2.6.0 - Default MK3S Printer profile
Octoprint 1.9.1
OctoPi 0.18.0
Octoeverywhere 2.7.1

I have Telegram notifications set up for Print Started, Print Complete, Print Failed, Print Time Elapsed every hour, First Layer Done, Filament or Color change, User Interaction Required, Print paused and Print Error. However I am only receiving Print Failed, Print Complete, Print Error and User Interaction Required notifications. I am not receiving any of the other notifications.

pifiuk, 2023-07-11:
update: Tried using Orcaslicer and the notifications work now so I guess this is linked to the prusaslicer gcode.

QuinnDamerell, 2023-07-14:
Thanks for letting me know; that's strange. Can you open a support ticket using the following link so we can share private information and figure it out?

https://octoeverywhere.com/support

closed 2023-07-14 by QuinnDamerell