QuinnDamerell/OctoPrint-OctoEverywhere #47: Better support/documentation for secrets management

loopj, 2023-07-29:
The default installation of octoeverywhere on moonraker creates an `octoeverywhere.conf` containing the `printer_id` and `private_key` of a printer.

Although there is a nice warning for people to not share the `private_key`, many people backup/share their printer config folder on github and are [inadvertently leaking their `printer_id` and `private_key`](https://github.com/search?q=path%3A**%2Foctoeverywhere.conf+private_key&type=code&ref=advsearch).

Moonraker has [support for storing secrets in a separate file](https://moonraker.readthedocs.io/en/latest/configuration/#secrets), so this could be a good approach for moonraker/klipper users.

QuinnDamerell, 2023-08-08:
I realized I never left a comment here. Thanks for bringing this up! We discussed it on our discord and we have an easy fix. We will use the moonraker secrets system in an update soon!

QuinnDamerell, 2023-08-15:
@loopj I'm looking into this now. Since the secrets file is owned by moonraker, I'm concerned about its usage. I'm thinking I might make my own .secrets file in the `<printer data>/octoeverywhere-store` folder, which is local storage and is not intended to be backed up.

What do you think?

QuinnDamerell, 2023-08-19:
This is done and deployed!

closed 2023-08-19 by QuinnDamerell