QuinnDamerell/OctoPrint-OctoEverywhere #49: Error: WebcamHelper _GetWebcamSettingObj exception

PseudoNinja, 2023-09-23:
### Description of Issue
During a 10MB print the printer stops printing but continued to heat bed and hot end. Display on printer says its still printing. Octopi shows an error that it lost communication with the printer and the only errors included in the log file is a repeating pattern of this error. 

Camera is plugged in BUT only displays a black screen. 

### My Hypothosis
My guess is its creating a memory leak by not properly collecting the object when the service call is over overflowing the onboard memory of the pi board rendering it unresponsive.

Thinking a quick null reference check could prevent this cascade.

### Configuration
- Ender 3 Pro (4.2.7 MB, Marlin2.0.1-V1.1.2-BLTouch-TMC2225 firmware)
- Raspberry PI 4 (Octoprint 1.9.2, OctoEverywhere 2.7.10)
- USB Webcam (Logitech C920)
- Fresh install of Octopi and first time running OctoEverywhere

### Octoprint.log file
```
2023-09-23 14:28:07,666 - octoprint.plugins.octoeverywhere - ERROR - WebcamHelper _GetWebcamSettingObj exception.; AttributeError Exception: 'NoneType' object has no attribute 'lower'; Traceback (most recent call last):
  File "/home/CE3P-A_admin/oprint/lib/python3.9/site-packages/octoeverywhere/webcamhelper.py", line 330, in _GetWebcamSettingObj
    a = self.WebcamPlatformHelperInterface.GetWebcamConfig()
  File "/home/CE3P-A_admin/oprint/lib/python3.9/site-packages/octoprint_octoeverywhere/octoprintwebcamhelper.py", line 54, in GetWebcamConfig
    cameraStreamerJmpegUrl = WebcamHelper.DetectCameraStreamerWebRTCStreamUrlAndTranslate(streamUrl)
  File "/home/CE3P-A_admin/oprint/lib/python3.9/site-packages/octoeverywhere/webcamhelper.py", line 382, in DetectCameraStreamerWebRTCStreamUrlAndTranslate
    streamUrlLower = streamUrl.lower()
AttributeError: 'NoneType' object has no attribute 'lower'
```

PseudoNinja, 2023-09-25:
E

QuinnDamerell, 2023-09-26:
Thanks for letting me know; I just fixed this in the most recent release!

closed 2023-09-26 by QuinnDamerell