QuinnDamerell/OctoPrint-OctoEverywhere #50: Not started

edombek, 2023-10-15:
```/home/orangepi/octoeverywhere-env/bin/python3 -m moonraker_octoeverywhere "token"```:
```Loading Service Config: {"KlipperConfigFolder": "/home/orangepi/printer_data/config", "MoonrakerConfigFile": "/home/orangepi/printer_data/config/moonraker.conf", "KlipperLogFolder": "/home/orangepi/printer_data/logs", "LocalFileStoragePath": "/home/orangepi/printer_data/octoeverywhere-store", "ServiceName": "octoeverywhere", "VirtualEnvPath": "/home/orangepi/octoeverywhere-env", "RepoRootFolder": "/home/orangepi/octoeverywhere"}
ERROR! - Exception while parsing service config. 'IsObserver'```

QuinnDamerell, 2023-10-16:
Thanks for letting me know! I just pushed a new version (2.7.34) that should fix this. You can update the repo manually or you can use the Mainsail / Fluidd UI to force an update check and install it. 

Let me know if it fixes the issue for you!

closed 2023-10-16 by edombek

QuinnDamerell, 2023-10-16:
No problem; thanks for letting me know!