QuinnDamerell/OctoPrint-OctoEverywhere #51: OctoEverywhere constantly writing errors to logs when klipper is shutdown by moonraker

Snelso91, 2023-11-11:
Following the docs from https://moonraker.readthedocs.io/en/latest/configuration/ and https://docs.mainsail.xyz/overview/quicktips/printer-power-switch, I have setup a `[power]` section in my moonraker.conf, which controls a mosfet switch via gpio which cuts power to the printer mainboard:

```
[power printer]
type: gpio
pin: gpio23
initial_state: off
off_when_shutdown: true
on_when_job_queued: true
restart_klipper_when_powered: true
bound_services:
 klipper
 klipper-mcu
```

Turning on the associated `printer` switch in the Mainsail Power Devices menu will start the bound services and also restart klipper when it is ready due to `restart_klipper_when_powered: true`. Similarly, turning off the printer switch will stop the bound services (klipper and klipper-mcu), which also triggers Mainsail to show a "Printer is powered off" section.
[image]

Despite this being intended behaviour for moonraker/mainsail, OctoEverywhere gets stuck in a loop constantly filling both the octoeverywhere.log and the /var/log/syslog every 2 seconds with messages of the type:
```
2023-11-11 14:54:05,536 - INFO - Connecting to moonraker: ws://0.0.0.0:7125/websocket
2023-11-11 14:54:05,554 - INFO - Websocket connected
2023-11-11 14:54:05,557 - INFO - Moonraker client websocket opened.
2023-11-11 14:54:05,560 - INFO - Moonraker client waiting for klippy ready...
2023-11-11 14:54:05,561 - INFO - Authenticating with moonraker...
2023-11-11 14:54:05,608 - INFO - Moonraker client waiting for klippy ready...
2023-11-11 14:54:05,618 - ERROR - Moonraker client is in an unknown klippy waiting state. state 'disconnected'
2023-11-11 14:54:05,620 - ERROR - Moonraker client exception in klippy waiting logic.; Exception Exception: Unknown klippy waiting state. disconnected; Traceback (most recent call last):
  File "/home/user/octoeverywhere/moonraker_octoeverywhere/moonrakerclient.py", line 584, in _AfterOpenReadyWaiter
    raise Exception(f"Unknown klippy waiting state. {state}")
Exception: Unknown klippy waiting state. disconnected

2023-11-11 14:54:05,621 - INFO - Moonraker client websocket shutdown called.
2023-11-11 14:54:05,626 - INFO - Moonraker client websocket shutdown complete.
2023-11-11 14:54:05,628 - INFO - Moonraker websocket connection closed.
2023-11-11 14:54:05,633 - INFO - Moonraker client websocket connection lost. We will try to restart it soon.
```

This causes the logs to rapidly increase in size over time as they are constantly being written to with these errors.
[octoeverywhere.log](https://github.com/QuinnDamerell/OctoPrint-OctoEverywhere/files/13325806/octoeverywhere.log)

QuinnDamerell, 2024-01-21:
Thanks for letting us know! We have made some changes to fix that, and we also have a limit to our log file size, so they can't grow too large. Let us know if it helps!

closed 2024-01-21 by QuinnDamerell

Snelso91, 2024-05-26:
> Thanks for letting us know! We have made some changes to fix that, and we also have a limit to our log file size, so they can't grow too large. Let us know if it helps!

I checked it out recently but forgot to get back to you, it seems that it is indeed fixed now!
Thanks for your help :)

Snelso91, 2024-05-26:
@QuinnDamerell I checked it out recently but forgot to get back to you, it seems that it is indeed fixed now!
Thanks for your help 🙂

QuinnDamerell, 2024-05-29:
Perfect, thanks for following up!