QuinnDamerell/OctoPrint-OctoEverywhere #58: [Feature Request] Bypass user input on install

jhaydter, 2024-05-04:
Currently install relies on user input to generate configuration for Bambu Lab Printer (SN, Access Code), not allowing automation in cases such as Dockerfile build, silent install, etc.

I'd like the ability to pass all configuration bypassing all input. 

One workaround I'm investigating is constructing the configuration files pre-install, to not trigger a new setup.

```
Stack Trace:
Traceback (most recent call last):
File "/home/ubuntu/octoeverywhere/py_installer/Installer.py", line 23, in Run
self._RunInternal()
File "/home/ubuntu/octoeverywhere/py_installer/Installer.py", line 134, in _RunInternal
configure.Run(context)
File "/home/ubuntu/octoeverywhere/py_installer/Configure.py", line 137, in Run
bc.EnsureBambuConnection(context)
File "/home/ubuntu/octoeverywhere/py_installer/NetworkConnectors/BambuConnector.py", line 37, in EnsureBambuConnection
ipOrHostname, port, accessToken, printerSn = self._SetupNewBambuConnection()
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/ubuntu/octoeverywhere/py_installer/NetworkConnectors/BambuConnector.py", line 78, in _SetupNewBambuConnection
accessCode = input("Enter your printer's Access Code: ")
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
EOFError: EOF when reading a line
```

QuinnDamerell, 2024-05-09:
This is a really interesting idea; I will try to add it as I add the docker support. Thanks for the suggestion!

QuinnDamerell, 2024-09-25:
I moved this to my TODO list, so I'm closing this issue for now!

closed 2024-09-25 by QuinnDamerell