QuinnDamerell/OctoPrint-OctoEverywhere #60: Installer fail

Shorezino, 2024-05-11:
Hello thank you for this service first of all. Issues downloading and other plugins on ender 3 v3 ke with new firmware. Seems like PIP issue. Days unable to fetch url and no suitable version available. This also happens when trying to install any other plugin like moon raker time lapse or Mobileraker companion. Then it shows installed but when I try to remove it I get the same error. Let me know if I can help or test anything. Thank you for this service.

Shorezino, 2024-05-11:
[image]

psiess, 2024-05-11:
The Octoeverywhere installer is failing for me on Creality K1 Max firmware V1.3.3.26. 

Running in K1 and K1 Max OS mode
Checking required system packages are installed...
System package install complete.
Checking Python Virtual Environment For OctoEverywhere...
No virtual environment found, creating one now.
Already using interpreter /usr/bin/python3
Using base prefix '/usr'
/usr/lib/python3.8/site-packages/virtualenv.py:1039: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
import imp
New python executable in /usr/data/octoeverywhere-env/bin/python3
Also creating executable in /usr/data/octoeverywhere-env/bin/python
Installing setuptools, pip, wheel...done.
Updating PIP if needed... (this can take a few seconds or so)
Could not fetch URL https://pypi.python.org/simple/pip/: There was a problem confirming the ssl certificate: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1108) - skipping
Requirement already up-to-date: pip in /usr/data/octoeverywhere-env/lib/python3.8/site-packages
Installing or updating required python libs...
Could not find a version that satisfies the requirement websocket_client<1.7.99,>=1.6.0 (from -r /usr/data/octoeverywhere/requirements.txt (line 10)) (from versions: )
No matching distribution found for websocket_client<1.7.99,>=1.6.0 (from -r /usr/data/octoeverywhere/requirements.txt (line 10))
2 replies 1 new
@[psiess](https://github.com/psiess)
psiess
[19 hours ago](https://github.com/QuinnDamerell/OctoPrint-OctoEverywhere/issues/60#discussioncomment-9390694)

jonny7737, 2024-05-11:
I have the same issue on my K1.

QuinnDamerell, 2024-05-12:
We are looking into this, it seems to be a problem with the date and time being set wrong. To work around it, try setting the date and time correctly.

Shorezino, 2024-05-12:
> We are looking into this, it seems to be a problem with the date and time being set wrong. To work around it, try setting the date and time correctly.

I tried the sudo date command from the install guide on octoeverywhere site for ender 3 v3 ke. date was right. Is this the method you're referring to? It seems to o be acting if an ssl cert is out of date

closed 2024-05-12 by QuinnDamerell

psiess, 2024-05-13:
There is still a problem.  The helper script won't install Octoeverywhere because it's already installed.  But, I can't remove Octoeverywhere because the removal fails with the pip certificate and websocket errors I reported earlier.  See my post from May 11.

jonny7737, 2024-05-13:
Only way to fix this is to reset via helper script and reinstall everything.

psiess, 2024-05-13:
> Only way to fix this is to reset via helper script and reinstall everything.

Ok thanks.  I'll do that.

QuinnDamerell, 2024-05-14:
@jonny7737 thanks for that tip! I will pass that along to others in the community.

psiess, 2024-05-14:
> Only way to fix this is to reset via helper script and reinstall everything.

Confirmed.  After factory reset and re-enabling root access, I was able to install the components I need.
Thanks again!