QuinnDamerell/OctoPrint-OctoEverywhere #63: Create bambu connect docker container as part as release process

ForceConstant, 2024-05-17:
Now that a couple of people have made example docker containers for the bambu connect, can you incorporate this into the main project, such that it also pushes to container repo when you make new releases in order to facilitate easy updating?

Thanks.

https://github.com/ntebis/octoeverywhere_bambulab_docker
https://github.com/RyanEwen/octoeverywhere_bambu_docker

QuinnDamerell, 2024-05-19:
I almost have it done! Check out this page for a quick overview, and let me know what you think!

https://github.com/QuinnDamerell/OctoPrint-OctoEverywhere/blob/docker/docker-readme.md

closed 2024-05-19 by QuinnDamerell

ForceConstant, 2024-05-19:
I am getting this over and over with this new container.

`2024-05-19 19:20:00,124 - INFO - Connection to the Bambu printer established! - Subscribing to the report subscription.
2024-05-19 19:20:00,130 - WARNING - Bambu printer connection lost. We will try to reconnect in a few seconds.
`

QuinnDamerell, 2024-05-19:
Can you double check the access code, sn, and up are correct?

ForceConstant, 2024-05-20:
Thanks, yes I had a typo in the Serial, now working.

QuinnDamerell, 2024-05-20:
Great! I will try to add some logic that will detect that and output a better error message.