QuinnDamerell/OctoPrint-OctoEverywhere #72: Alternative Cam view not refreshed when replying to "Are You Still There?" message

edstewbob, 2024-07-01:
Using Bambu A1 Mini every 2 minutes on the Quick View screen a message appears that asks "Are You Still There?" requiring a reply to continue viewing. When responding to this message and using the Alternative Cam, the Quick View camera view shows a small question mark in the center of the screen and only shows an image if clicking on Bambu Cam and then back to the Alternative Cam view to see the active print.
[image]

QuinnDamerell, 2024-08-11:
Thanks for reporting that; I haven't seen that before! When the image is broken, if you refresh the page, does the webcam start working again?

QuinnDamerell, 2024-08-17:
Re-open if this still repos!

closed 2024-08-17 by QuinnDamerell