QuinnDamerell/OctoPrint-OctoEverywhere #73: unable to install Neptune 4 Max

Resinato, 2024-08-10:
Hi, I followed the guide at https://blog.octoeverywhere.com/elegoo-neptune-4-series-free-remote-access-and-ai-failure-detection/?source=website_printer_select

The printer is updated with the latest Elegoo firmware (as for 2024-08-10)

I updated the apt lists (editing them to "archive.debian.org") since I had an error:
Err:6 http://deb.debian.org/debian buster-backports Release
  404  Not Found [IP: 146.75.54.132 80]

I am still stuck here:
Checking Python Virtual Environment For OctoEverywhere...
Updating PIP if needed... (this can take a few seconds or so)
/home/mks/octoeverywhere-env/bin/python: No module named pip

any idea?

QuinnDamerell, 2024-08-11:
Thanks for reaching out! That looks like it might be a local issue; it's an error in the system package updater; it's saying it can't find a package on their servers. Can you try running `sudo apt-get update` and then `sudo apt-get upgrade` and see if it resolves it?

Resinato, 2024-08-12:
Hi there,
i run the update and upgrade commands. 
I have a couple of things I can't upgrade:
linux-dtb-edge-rockchip64/buster 24.5.1 arm64 [upgradable from: 22.05.0-trunk]
linux-image-edge-rockchip64/buster 24.5.1 arm64 [upgradable from: 22.05.0-trunk]

here you have the output, same problem as yesterday:

`bash <(curl -s https://octoeverywhere.com/install.sh)
You might be asked for your system password - this is required to install the required system packages.
Hit:1 http://security.debian.org buster/updates InRelease
Hit:2 http://archive.debian.org/debian buster InRelease
Hit:3 http://archive.debian.org/debian buster-updates InRelease
Hit:4 http://archive.debian.org/debian buster-backports InRelease
Get:6 https://pkgs.tailscale.com/stable/debian bookworm InRelease
Hit:5 http://netcup.armbian.com/apt buster InRelease
Fetched 6,578 B in 3s (1,977 B/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
2 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree
Reading state information... Done
git is already the newest version (1:2.20.1-2+deb10u9).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
MKS-PI detected.
Mon 12 Aug 2024 11:08:06 PM CST
fatal: not in a git directory
OctoEverywhere repo already exists, skipping clone.
Latest git tag found 3.4.5, current status 2.10.8-115-g9de7179

      @@@@@@@@@@@@@@@@@@@@@@@@***@@@@@@@@@@@@@@@@@@@@@@@
      @@@@@@@@@@@@@@***********************@@@@@@@@@@@@@
      @@@@@@@@@@*******************************@@@@@@@@@
      @@@@@@@@***********************************@@@@@@@
      @@@@@,,,************************/////////*****@@@@
      @@@@,,,,,,*****************//////////////******@@@
      @@,,,,,,,,,,***********//////////////////*******@@
      @@,,,,,,,,,,,,*******////////****///////*********@
      @,,,,,,,,,,,/////////////////****//////***********
      @,,,,,,,//////////////////////////////************
      ,,,,,,,,////////////////////////////**************
      @,,,,,,,,,,,,/////////////////////****************
      @,,,,,,,,,,,,,,/////////////////******************
      @@,,,,,,,,,,,,,,,,//////////////*****************@
      @@@,,,,,/#######,,,,///////////*****************@@
      @@@@,,,##########,,,,,,,//////,****************@@@
      @@@@@,##########,,,,,,,,,////,,,,*************@@@@
      @@@@@########,,,,,,,,,,,,//,,,,,,,,********@@@@@@@
      @@@@@#@@@@,,,,,,,,,,,,,,,,,,,,,,,,,,,***,@@@@@@@@@
      @@@@@@@@@@@@@@@,,,,,,,,,,,,,,,,,,,,,@@@@@@@@@@@@@@

  OctoEverywhere For Klipper, Creality OS, And Bambu Lab Printers
 The 3D Printing Communities #1 Remote Access And AI Cloud Service

OctoEverywhere empowers the worldwide maker community with...
  - Free & Unlimited Mainsail, Fluidd, And Bambu Lab Printers Remote Access
  - Free & Unlimited Next-Gen AI Print Failure Detection
  - Free Full Frame Rate & Full Resolution Webcam Streaming
  - 5 Star Rated iOS & Android Apps
  - Real-Time Print Notifications
  - And So Much More

Checking required system packages are installed...
Installing required system packages...
You might be asked for your system password - this is required to install the required system packages.
Mon 12 Aug 2024 11:08:08 PM CST
Reading package lists... Done
Building dependency tree
Reading state information... Done
python3-pip is already the newest version (18.1-5).
virtualenv is already the newest version (15.1.0+ds-2+deb10u1).
python3 is already the newest version (3.7.3-1).
python3-venv is already the newest version (3.7.3-1).
curl is already the newest version (7.64.0-4+deb10u9).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
Ensuring zlib is install for Pillow, it's ok if this package install fails.
Reading package lists... Done
Building dependency tree
Reading state information... Done
zlib1g-dev is already the newest version (1:1.2.11.dfsg-1+deb10u2).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
Reading package lists... Done
Building dependency tree
Reading state information... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package python-imaging is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  python-pil:armhf python-pil

Reading package lists... Done
Building dependency tree
Reading state information... Done
python3-pil is already the newest version (5.4.1-2+deb10u6).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'python3-pil' instead of 'python3-pillow'
python3-pil is already the newest version (5.4.1-2+deb10u6).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
Checking Python Virtual Environment For OctoEverywhere...
Updating PIP if needed... (this can take a few seconds or so)
/home/mks/octoeverywhere-env/bin/python: No module named pip
`

QuinnDamerell, 2024-08-17:
That's strange. Can you delete the `/home/mks/octoeverywhere-env` folder and run the installer again?

QuinnDamerell, 2024-09-25:
Closing this for now, please re-open if you are still having issues!

closed 2024-09-25 by QuinnDamerell