QuinnDamerell/OctoPrint-OctoEverywhere #76: Unable to run as different uid because /root/octoeverywhere-env/bin/python is not readable by non root

halkeye, 2024-10-25:
Default runs normally
```
docker run -it --rm octoeverywhere/octoeverywhere:3.6.1
2024-10-25 21:39:15,768 - INFO - ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2024-10-25 21:39:15,768 - INFO - Starting Docker OctoEverywhere Bootstrap
2024-10-25 21:39:15,768 - INFO - ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2024-10-25 21:39:15,768 - INFO - Env Vars: environ({'PATH': '/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin', 'HOSTNAME': '5028a32ee1ca', 'TERM': 'xterm', 'USER': 'root', 'REPO_DIR': '/root/octoeverywhere', 'VENV_DIR': '/root/octoeverywhere-env', 'DATA_DIR': '/data/', 'HOME': '/root'})
2024-10-25 21:39:15,768 - INFO - Ensuring path exists: /root/octoeverywhere-env
2024-10-25 21:39:15,768 - INFO - Ensuring path exists: /root/octoeverywhere
2024-10-25 21:39:15,768 - INFO - Ensuring path exists: /data/
2024-10-25 21:39:15,769 - ERROR - Exception while bootstrapping up OctoEverywhere Bambu Connect.; Exception Exception: Path does not exist: /data/; Traceback (most recent call last):
  File "/root/octoeverywhere/docker_octoeverywhere/__main__.py", line 65, in <module>
    dataPath = EnsureIsPath(os.environ.get("DATA_DIR", None))
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/octoeverywhere/docker_octoeverywhere/__main__.py", line 53, in EnsureIsPath
    raise Exception(f"Path does not exist: {path}")
Exception: Path does not exist: /data/
```

but fails with custom uuid
```
docker run -it --rm -u 1000:1000 octoeverywhere/octoeverywhere:3.6.1
/root/octoeverywhere-env/bin/python: No module named docker_octoeverywhere
```

But make /root readable and it'll work just fine

```
cat Dockerfile
FROM octoeverywhere/octoeverywhere:3.6.1

RUN chmod a+rx /root /root/octoeverywhere-env /root/octoeverywhere-env/bin /root/octoeverywhere-env/bin/python
```

```
docker run -it --rm -u 1000:1000 halkeye bash
2024-10-25 21:43:25,091 - INFO - ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2024-10-25 21:43:25,091 - INFO - Starting Docker OctoEverywhere Bootstrap
2024-10-25 21:43:25,091 - INFO - ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2024-10-25 21:43:25,091 - INFO - Env Vars: environ({'PATH': '/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin', 'HOSTNAME': '03a91c553b3c', 'TERM': 'xterm', 'USER': 'root', 'REPO_DIR': '/root/octoeverywhere', 'VENV_DIR': '/root/octoeverywhere-env', 'DATA_DIR': '/data/', 'HOME': '/'})
2024-10-25 21:43:25,091 - INFO - Ensuring path exists: /root/octoeverywhere-env
2024-10-25 21:43:25,092 - INFO - Ensuring path exists: /root/octoeverywhere
2024-10-25 21:43:25,092 - INFO - Ensuring path exists: /data/
2024-10-25 21:43:25,093 - ERROR - Exception while bootstrapping up OctoEverywhere Bambu Connect.; Exception Exception: Path does not exist: /data/; Traceback (most recent call last):
  File "/root/octoeverywhere/docker_octoeverywhere/__main__.py", line 65, in <module>
    dataPath = EnsureIsPath(os.environ.get("DATA_DIR", None))
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/octoeverywhere/docker_octoeverywhere/__main__.py", line 53, in EnsureIsPath
    raise Exception(f"Path does not exist: {path}")
Exception: Path does not exist: /data/
```

My vote is to deploy to anywhere other than /root, but chmod would work too.

QuinnDamerell, 2024-11-20:
Thanks for letting me know! Ideally, you don't want to install the plugin as the root user because then stuff like this happens. Can you try installing the user as a non-root user and see if it works?

closed 2024-11-20 by QuinnDamerell

halkeye, 2024-11-20:
you should be able to test with:
```
docker run -it --rm -u 1000:1000 octoeverywhere/octoeverywhere:3.6.1
/root/octoeverywhere-env/bin/python: No module named docker_octoeverywhere
```