QuinnDamerell/OctoPrint-OctoEverywhere #77: Unable to login to bambu cloud

juanfranciscolopez28, 2024-10-29:
Hi!

Since a few days ago a 403 error occurs when trying to connect to bambu cloud, the error that appears in the console is the following:

```
2024-10-29 20:00:31,168 - ERROR - Login Bambu Cloud failed with status code: 403, Response: 

2024-10-29 20:00:31,170 - ERROR - ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

2024-10-29 20:00:31,170 - ERROR -                                                      Failed To Log Into Bambu Cloud

2024-10-29 20:00:31,171 - ERROR - Unknown error, we will try again later.
```

I use docker compose with the default configuration that appears in the wiki. Until a few days ago it worked correctly.

Best regards and thanks!

QuinnDamerell, 2024-11-04:
Thanks for the bug report! I will look into this.

If you can, I would recommend you use the LAN connection for Bambu Connect, it's much more stable.

ados8, 2024-11-04:
Just discovered OctoEverywhere and this issue was my first main sticking point.
I don't want to enable LAN mode to get it working to then loose mobile and remote control.
Hoping the bug is sorted soon. 😊

ados8, 2024-11-05:
Found if you keep setting LAN only mode and cloud mode multiple times you eventually glitch it to work in cloud mode.
Makes it much easier to print as you can use the mobile app and Bambu Studio doesn't require selecting the printer every time.

shatteredeagle, 2024-11-12:
Getting the same error. @ados8 where were you switching the modes back and forth on the printer or the app config (or both :-) )

Rulevoid, 2024-11-20:
I had a working config with an A1 and an A1m without LAN only, then rebuilt my docker containers and now get this same issue. I can no longer login and see the same error even after turning off MFA and confirming credentials.

Any update on this?

TheToby1, 2024-12-19:
Having the same issue.

QuinnDamerell, 2024-12-19:
Can you open a support ticket and send me your logs?

But keep in mind that the Bambu Cloud config isn't great. Bambu has made the APIs hard to work with for 3rd party app devs, and they keep breaking the API. I recommend you use the local connection mode. You **do not need** to enable LAN Only mode on your printer, it should work with it disabled.

michalsteyn, 2024-12-19:
Hi @QuinnDamerell,
Re: `You do not need to enable LAN Only mode on your printer, it should work with it disabled.` That is VERY interesting... Don't think that is documented anywhere.

I just tried this, and it seems when configuring the docker container to use LAN Only, it still tries to connect to the cloud.
Here is my docker compose file:
```
  octoeverywhere-bambu-connect:
    # https://octoeverywhere.com/s/bambu-setup
    image: octoeverywhere/octoeverywhere:latest
    environment:
      - SERIAL_NUMBER=###REDACTED###
      - PRINTER_IP=###REDACTED###
      - ACCESS_CODE=###REDACTED###
      - LAN_ONLY_MODE=TRUE
    volumes:
      - ./bambulab:/data
```

Here are my logs:

```
2024-12-19 21:46:14,267 - INFO - ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2024-12-19 21:46:14,267 - INFO - Starting Docker OctoEverywhere Bootstrap
2024-12-19 21:46:14,267 - INFO - ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2024-12-19 21:46:14,267 - INFO - Env Vars: environ({'PATH': '/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin', 'HOSTNAME': '###REDACTED###', 'SERIAL_NUMBER': '###REDACTED###', 'PRINTER_IP': '###REDACTED###', 'ACCESS_CODE': '###REDACTED###', 'LAN_ONLY_MODE': 'TRUE', 'USER': 'root', 'REPO_DIR': '/root/octoeverywhere', 'VENV_DIR': '/root/octoeverywhere-env', 'DATA_DIR': '/data/', 'HOME': '/root'})
2024-12-19 21:46:14,267 - INFO - Ensuring path exists: /root/octoeverywhere-env
2024-12-19 21:46:14,267 - INFO - Ensuring path exists: /root/octoeverywhere
2024-12-19 21:46:14,267 - INFO - Ensuring path exists: /data/
2024-12-19 21:46:14,267 - INFO - Init config object: /data/
2024-12-19 21:46:14,268 - INFO - Setting Serial Number: ###REDACTED###
2024-12-19 21:46:14,271 - INFO - Connection Mode: LAN Only (Use the env var LAN_ONLY_MODE=FALSE to enable Bambu Cloud mode.)
2024-12-19 21:46:14,271 - INFO - Setting Access Code: ###REDACTED###
2024-12-19 21:46:14,272 - INFO - Setting Printer IP: ###REDACTED###
2024-12-19 21:46:14,273 - INFO - Launch config: {"ServiceName": "octoeverywhere", "CompanionInstanceIdStr": "1", "VirtualEnvPath": "/root/octoeverywhere-env", "RepoRootFolder": "/root/octoeverywhere", "LocalFileStoragePath": "/data/octoeverywhere-store", "LogFolder": "/data/logs", "ConfigFolder": "/data/"}
2024-12-19 21:46:14,273 - INFO - Launch PY path: /root/octoeverywhere-env/bin/python3
Loading Service Config: {"ServiceName": "octoeverywhere", "CompanionInstanceIdStr": "1", "VirtualEnvPath": "/root/octoeverywhere-env", "RepoRootFolder": "/root/octoeverywhere", "LocalFileStoragePath": "/data/octoeverywhere-store", "LogFolder": "/data/logs", "ConfigFolder": "/data/"}
2024-12-19 21:46:14,823 - INFO - ################################################
2024-12-19 21:46:14,823 - INFO - #### OctoEverywhere Bambu Connect Starting #####
2024-12-19 21:46:14,824 - INFO - ################################################
2024-12-19 21:46:14,824 - INFO - Plugin Version: 3.6.3
2024-12-19 21:46:14,957 - INFO - ZStandard Dict Training loaded. Data Length:112640 DictID:1
2024-12-19 21:46:14,957 - INFO - Compression is using zstandard with 2 threads
2024-12-19 21:46:14,958 - INFO - mDns Cache file loaded. Cached Entries Found: 0
2024-12-19 21:46:14,959 - INFO - OctoPingPong stats loaded from file.
2024-12-19 21:46:14,959 - INFO - PingPong has detected a first time run. Updating latency stats now.
2024-12-19 21:46:14,961 - INFO - Attempting to talk to OctoEverywhere, server con wss://starport-v1.octoeverywhere.com/octoclientws[1] wsId:133702948003312
2024-12-19 21:46:14,962 - INFO - Logging into Bambu Cloud...
2024-12-19 21:46:15,084 - ERROR - Login Bambu Cloud failed with status code: 403, Response: 
2024-12-19 21:46:15,085 - ERROR - ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2024-12-19 21:46:15,085 - ERROR -                                                      Failed To Log Into Bambu Cloud
2024-12-19 21:46:15,085 - ERROR - Unknown error, we will try again later.
2024-12-19 21:46:15,086 - ERROR - ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2024-12-19 21:46:15,147 - INFO - Websocket connected
2024-12-19 21:46:15,148 - INFO - Connected To OctoEverywhere, server con wss://starport-v1.octoeverywhere.com/octoclientws[1]. Starting handshake...
2024-12-19 21:46:15,175 - WARNING - Found device ID not found on platform: linux
2024-12-19 21:46:15,335 - INFO - Handshake complete, server con wss://starport-v1.octoeverywhere.com/octoclientws[1], successfully connected to OctoEverywhere!
2024-12-19 21:46:15,335 - INFO - Primary Connection To OctoEverywhere Established - We Are Ready To Go!
2024-12-19 21:47:06,609 - INFO - Web Stream 3 tried to send a message after close.
2024-12-19 21:47:14,331 - INFO - Web Stream 6 tried to send a message after close.
2024-12-19 21:47:16,607 - WARNING - BambuWebcamHelper wasn't able to get the printer state after 6 attempts
2024-12-19 21:47:16,608 - INFO - QuickCam capture thread starting.
2024-12-19 21:47:18,502 - INFO - Web Stream 2 tried to send a message after close.
2024-12-19 21:47:19,974 - WARNING - BambuWebcamHelper wasn't able to get the printer state after 6 attempts
2024-12-19 21:47:19,975 - INFO - Web Stream 4 tried to send a message after close.
2024-12-19 21:47:20,341 - INFO - Web Stream 7 tried to send a message after close.
[... cut ...]
2024-12-19 21:48:28,443 - INFO - Web Stream 40 tried to send a message after close.
2024-12-19 21:48:33,444 - INFO - Web Stream 41 tried to send a message after close.
2024-12-19 21:48:33,447 - INFO - Web Stream 42 tried to send a message after close.
2024-12-19 21:48:38,448 - INFO - Web Stream 44 tried to send a message after close.
```

On the OctoEverywhere Dashboard I can see a video stream of my printer, but it also shows a Red Error: `Bambu Connection Lost` and it doesn't have any info about the print job.

[image]

QuinnDamerell, 2024-12-19:
@michalsteyn thanks for the logs! Yeah, the 403 probs means that Bambu's login requires an email two-factor code, which they don't allow 3rd party devs to implement right now. :(

Try the local connection mode. I'm updating the docs right now to make that mode more clear. Let me know if it works for you!

Rulevoid, 2024-12-19:
Is the 2FA being enforced on API only? Because I disabled 2FA at least for
the web portal login. I'll try the local mode whne you've updated the docs.
Thanks!

QuinnDamerell, 2024-12-19:
Yeah, it's not 2FA; it's an email code-based challenge. The backend randomly requires it, but I have gotten in a few times by logging into the website normally.

I just updated the blog posts with new info about using the local connection mode.

michalsteyn, 2024-12-19:
Thanks for the quick response @QuinnDamerell.

Just to be clear, in my config I set: `LAN_ONLY_MODE=TRUE`, why is it still trying to connect to the cloud? Did I miss something, or is this a bug you have to fix first?

Just want to double check if I have to change my config, or just wait for an update.

Super thanks for your hard work!

QuinnDamerell, 2024-12-20:
I just pushed an update, so you will want to pull the newest docker container. I updated the docker compose file, you can find the new version in the repo root. Your current version should still work as well, but the new one is cleaner and more clear (I hope). If you want to make sure you're using the local connection mode, you can use `CONNECTION_MODE` var by setting it to `local`. 

Hopefully that helps! Let me know if not.

closed 2024-12-20 by QuinnDamerell

Rulevoid, 2024-12-20:
I pulled latest and added that var, but Bambu Cloud is still picked up for connection mode resulting in the same problem. I'm on Windows Docker.

```
2024-12-19 21:22:47 2024-12-20 02:22:47,045 - INFO - ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2024-12-19 21:22:47 2024-12-20 02:22:47,045 - INFO - Starting Docker OctoEverywhere Bootstrap
2024-12-19 21:22:47 2024-12-20 02:22:47,045 - INFO - ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2024-12-19 21:22:47 2024-12-20 02:22:47,045 - INFO - Env Vars: environ({'PATH': '/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin', 'HOSTNAME': '748d2b14d4ea', 'ACCESS_CODE': 'xxx', 'LAN_ONLY_MODE': 'FALSE', 'CONNECTION_MODE': 'local', 'SERIAL_NUMBER': 'xxx', 'PRINTER_IP': '192.168.1.123', 'BAMBU_CLOUD_ACCOUNT_EMAIL': 'xxx@gmail.com', 'BAMBU_CLOUD_ACCOUNT_PASSWORD': 'xxx', 'USER': 'root', 'REPO_DIR': '/root/octoeverywhere', 'VENV_DIR': '/root/octoeverywhere-env', 'DATA_DIR': '/data/', 'HOME': '/root'})
2024-12-19 21:22:47 2024-12-20 02:22:47,045 - INFO - Ensuring path exists: /root/octoeverywhere-env
2024-12-19 21:22:47 2024-12-20 02:22:47,045 - INFO - Ensuring path exists: /root/octoeverywhere
2024-12-19 21:22:47 2024-12-20 02:22:47,045 - INFO - Ensuring path exists: /data/
2024-12-19 21:22:47 2024-12-20 02:22:47,045 - INFO - Init config object: /data/
2024-12-19 21:22:47 2024-12-20 02:22:47,047 - INFO - Setting Serial Number: 03919C450801736
2024-12-19 21:22:47 2024-12-20 02:22:47,051 - INFO - Connection Mode: Bambu Cloud (Use the env var LAN_ONLY_MODE=TRUE to enable LAN Only mode.)
```

QuinnDamerell, 2024-12-21:
It looks like that's not the latest container, it's `3.6.3`. Can you try pulling the container again?

Rulevoid, 2024-12-21:
Thanks for catching that! Just pulled and it's all working again. Cheers!

QuinnDamerell, 2024-12-21:
No problem! I’m really glad to hear it’s working! Let me know if you hit any other issues.

- Quinn