QuinnDamerell/OctoPrint-OctoEverywhere #78: unable to access 'https://github.com/QuinnDamerell/OctoPrint-OctoEverywhere.git/': Could not resolve host: github.com

jamesonvera, 2024-10-31:
fatal: unable to access 'https://github.com/QuinnDamerell/OctoPrint-OctoEverywhere.git/': Could not resolve host: github.com

#=======================================================#
 Cloning OctoEverywhere from
 https://github.com/QuinnDamerell/OctoPrint-OctoEverywhere.git
 failed!

what can i do?

QuinnDamerell, 2024-11-04:
Hey! This is usually due to a local issue, the error is that the installer script can't do a DNS resolve of github.com. Can you make sure your device is able to access the internet and can access github?

closed 2024-11-04 by QuinnDamerell

jamesonvera, 2024-11-04:
> <QuinnDamerell/OctoPrint-OctoEverywhere/issue/78/issue_event/15097963898@

I have already verified that the devices can connect directly to the
internet but I cannot make a connection with git