QuinnDamerell/OctoPrint-OctoEverywhere #79: Webhook settings don't save

ados8, 2024-11-05:
Currently using trial while I test out the system.
Noticed Webhooks don't work because the settings never save.

Setup device printer notification -> webhooks -> URL
Set the URL, it will remain there until you leave to home screen and then go back, now it's empty.
Explains why I don't get notifications, happens with local and public requests.

ados8, 2024-11-09:
Any thoughts on this one, even with all the battery settings correct the notifications from the web app are majorly delayed.
I have a webhook server ready but can't get the settings to stay in the account.

QuinnDamerell, 2024-11-20:
Thanks for the feedback! The UI is currently setup to now show the existing URL, which is why it seems like it doesn't save. But I will work on it to see if I can make that better. I also want to add a button to send test notifications so it's easier to setup and debug.