QuinnDamerell/OctoPrint-OctoEverywhere #81: [FEATURE REQUEST] live stream expansion

ados8, 2024-11-08:
Currently the live stream could be expanded to offer more features.
Have a list of suggestions based on current use.

- New live stream option, 'Pause Stream When The Current Print Completes'. Currently it either runs always or ends when the print does. For those wanting more security i.e. secondary camera, having an option for the steam to pause with a message to wait until next live print. That way the URL doesn't need to be generated again and again per print while having more security than continuous. It would then automatically start the camera feed again once the next print begins.

- Live stream manager, a way of keeping track of current stream URLs and the option to change and delete them. Currently if you don't keep a copy of the stream URL there's no way to find it again, so you would want to hope you don't make it continuous.

- Security to live streams, simple password, this will prevent bots or random people from accessing your link.

- View counter so you can see how many are or max watched it. This will also help with security as you can delete a link if it has high views.

ados8, 2024-11-09:
Adding another to the list for security.
Can there be an option to have the camera stream completely disconnected even to the owners account when the printer isn't going.
Essentially when they view the printer and a job is in progress the camera says privacy mode.

ados8, 2025-01-09:
Closing as there has been no response.

closed 2025-01-09 by ados8

QuinnDamerell, 2025-01-10:
Hey, Sorry about that. I saw this and noted the ideas but didn't respond.