QuinnDamerell/OctoPrint-OctoEverywhere #84: [Feature Request] Support for multi-tool?

Martorias, 2024-12-02:
This is perhaps overkill but when printing with more than one toolhead only the first/original tool gets temp reported meaning it'll say 30c/off or similar for this entire print. 

Either if possible it would be cool if all tools were reported, or if the current active tools temp is reported.

Martorias, 2024-12-02:
Actually this might be somewhat of a bug report as well as I noticed the progress and timer doesn't work either :-)
It's still in status Warming Up after an hour.

QuinnDamerell, 2024-12-08:
Thanks for the idea, I will add it to the TODO list!

About the warming up, can you check your log to see what's going on? There's some reason that it thinks it's not printing yet.

closed 2024-12-08 by QuinnDamerell