QuinnDamerell/OctoPrint-OctoEverywhere #87: High CPU load when printer is offline

drawgas, 2024-12-20:
I've noticed that Bambu Lab connection is using quite some CPU when printer is turned off. Could this be somehow reduced with configuration?
Running OctoEverywhere 3.6.4 on Raspberry Pi 3.
Planning on adding another Bambu printer and worried that it will eat even more CPU when they both are offline.

[image]

boehser-enkel, 2025-01-14:
Same here.
Corresponding log: 

```
2025-01-14 05:55:37,207 - INFO - Searching for your Bambu Lab printer 0309DA3A2000673

2025-01-14 05:55:48,562 - INFO - Trying to connect to printer via local connection...

2025-01-14 05:55:49,068 - ERROR - Failed to connect to the Bambu printer None:8883. We will retry in a bit.; OSError Exception: [Errno 113] Host is unreachable; Traceback (most recent call last):

  File "/app/octoeverywhere/bambu_octoeverywhere/bambuclient.py", line 147, in _ClientWorker

    self.Client.connect(connectionContext.IpOrHostname, int(self.PortStr), keepalive=5)

  File "/app/octoeverywhere-env/lib/python3.12/site-packages/paho/mqtt/client.py", line 1435, in connect

    return self.reconnect()

           ^^^^^^^^^^^^^^^^

  File "/app/octoeverywhere-env/lib/python3.12/site-packages/paho/mqtt/client.py", line 1598, in reconnect

    self._sock = self._create_socket()

                 ^^^^^^^^^^^^^^^^^^^^^

  File "/app/octoeverywhere-env/lib/python3.12/site-packages/paho/mqtt/client.py", line 4609, in _create_socket

    sock = self._create_socket_connection()

           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/app/octoeverywhere-env/lib/python3.12/site-packages/paho/mqtt/client.py", line 4640, in _create_socket_connection

    return socket.create_connection(addr, timeout=self._connect_timeout, source_address=source)

           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/usr/lib/python3.12/socket.py", line 865, in create_connection

    raise exceptions[0]

  File "/usr/lib/python3.12/socket.py", line 850, in create_connection

    sock.connect(sa)

OSError: [Errno 113] Host is unreachable
```

QuinnDamerell, 2025-01-24:
Thanks for the feedback! I just fixed this in the main branch; it will roll out with the next update!

closed 2025-01-24 by QuinnDamerell