QuinnDamerell/OctoPrint-OctoEverywhere #88: Bambu Connection Lost (just started today)

RyanEwen, 2025-01-09:
I suspect Bambu has updated their cloud login process and is now blocking OctoEverywhere:

[image]

I noticed this first on an existing printer, and second on a brand new printer I was just adding, followed by the rest of my printers upon restarting the docker containers.

Logs from the new printer:

```bash
p1s_6-1  | 2025-01-09 17:11:56,168 - INFO - ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
p1s_6-1  | 2025-01-09 17:11:56,168 - INFO - Starting Docker OctoEverywhere Bootstrap
p1s_6-1  | 2025-01-09 17:11:56,168 - INFO - ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
p1s_6-1  | 2025-01-09 17:11:56,168 - INFO - Env Vars: environ({'PATH': '/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin', 'HOSTNAME': 'bc1605568b07', 'SERIAL_NUMBER': '01P09C490203723', 'ACCESS_CODE': '31486852', 'BAMBU_CLOUD_ACCOUNT_EMAIL': 'ryan.ewen@gmail.com', 'BAMBU_CLOUD_ACCOUNT_PASSWORD': 'BBL360rach!', 'PRINTER_IP': '192.168.2.164', 'USER': 'app', 'REPO_DIR': '/app/octoeverywhere', 'VENV_DIR': '/app/octoeverywhere-env', 'DATA_DIR': '/data/', 'HOME': '/root'})
p1s_6-1  | 2025-01-09 17:11:56,168 - INFO - Ensuring path exists: /app/octoeverywhere-env
p1s_6-1  | 2025-01-09 17:11:56,168 - INFO - Ensuring path exists: /app/octoeverywhere
p1s_6-1  | 2025-01-09 17:11:56,168 - INFO - Ensuring path exists: /data/
p1s_6-1  | 2025-01-09 17:11:56,169 - INFO - Init config object: /data/
p1s_6-1  | 2025-01-09 17:11:56,169 - INFO - Setting Serial Number: 01P09C490203723
p1s_6-1  | 2025-01-09 17:11:56,171 - INFO - Setting Access Code: 31486852
p1s_6-1  | 2025-01-09 17:11:56,171 - INFO - Setting Printer IP: 192.168.2.164
p1s_6-1  | 2025-01-09 17:11:56,172 - INFO - Bambu cloud email or password supplied, so we will use Bambu Cloud connection mode.
p1s_6-1  | 2025-01-09 17:11:56,172 - INFO - If you want to change the connection mode, use `CONNECTION_MODE` which can be set to 'cloud' or 'local'.
p1s_6-1  | 2025-01-09 17:11:56,173 - INFO - Setting Bambu Cloud Context: ryan.ewen@gmail.com
p1s_6-1  | 2025-01-09 17:11:56,174 - INFO - Setting Bambu Cloud to the default value for world wide accounts.
p1s_6-1  | 2025-01-09 17:11:56,174 - INFO - Launch config: {"ServiceName": "octoeverywhere", "CompanionInstanceIdStr": "1", "VirtualEnvPath": "/app/octoeverywhere-env", "RepoRootFolder": "/app/octoeverywhere", "LocalFileStoragePath": "/data/octoeverywhere-store", "LogFolder": "/data/logs", "ConfigFolder": "/data/"}
p1s_6-1  | 2025-01-09 17:11:56,174 - INFO - Launch PY path: /app/octoeverywhere-env/bin/python3
p1s_6-1  | Loading Service Config: {"ServiceName": "octoeverywhere", "CompanionInstanceIdStr": "1", "VirtualEnvPath": "/app/octoeverywhere-env", "RepoRootFolder": "/app/octoeverywhere", "LocalFileStoragePath": "/data/octoeverywhere-store", "LogFolder": "/data/logs", "ConfigFolder": "/data/"}
p1s_6-1  | 2025-01-09 17:11:56,658 - INFO - ################################################
p1s_6-1  | 2025-01-09 17:11:56,658 - INFO - #### OctoEverywhere Bambu Connect Starting #####
p1s_6-1  | 2025-01-09 17:11:56,658 - INFO - ################################################
p1s_6-1  | 2025-01-09 17:11:56,658 - INFO - Plugin Version: 3.6.5
p1s_6-1  | Secrets file doesn't exist. Creating a new file now!
p1s_6-1  | 2025-01-09 17:11:56,659 - INFO - No printer id was found, generating one now!
p1s_6-1  | 2025-01-09 17:11:56,659 - INFO - New printer id created: BZ2YA1X6NLIH5ZW9FS1CZYY9RQB5MI854W2E522A5D7MC94WB4Y4UKCTFA3P
p1s_6-1  | 2025-01-09 17:11:56,660 - INFO - No private key was found, generating one now!
p1s_6-1  | 2025-01-09 17:11:56,660 - INFO - New private key created.
p1s_6-1  | 2025-01-09 17:11:56,663 - INFO - ZStandard Dict Training loaded. Data Length:112640 DictID:1
p1s_6-1  | 2025-01-09 17:11:56,664 - INFO - Compression is using zstandard with 2 threads
p1s_6-1  | 2025-01-09 17:11:56,664 - INFO - MDns cache file reset
p1s_6-1  | 2025-01-09 17:11:56,664 - INFO - OctoPingPong stats reset
p1s_6-1  | 2025-01-09 17:11:56,665 - INFO - PingPong has detected a first time run. Updating latency stats now.
p1s_6-1  | 2025-01-09 17:11:56,666 - INFO - Attempting to talk to OctoEverywhere, server con wss://starport-v1.octoeverywhere.com/octoclientws[1] wsId:126956502439248
p1s_6-1  | 2025-01-09 17:11:56,667 - INFO - Logging into Bambu Cloud...
p1s_6-1  | 2025-01-09 17:11:56,710 - ERROR - Login Bambu Cloud failed with status code: 403, Response: 
p1s_6-1  | 2025-01-09 17:11:56,710 - ERROR - ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
p1s_6-1  | 2025-01-09 17:11:56,710 - ERROR -                                                      Failed To Log Into Bambu Cloud
p1s_6-1  | 2025-01-09 17:11:56,710 - ERROR - Unknown error, we will try again later.
p1s_6-1  | 2025-01-09 17:11:56,710 - ERROR - ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
p1s_6-1  | 2025-01-09 17:11:56,785 - INFO - Websocket connected
p1s_6-1  | 2025-01-09 17:11:56,786 - INFO - Connected To OctoEverywhere, server con wss://starport-v1.octoeverywhere.com/octoclientws[1]. Starting handshake...
p1s_6-1  | 2025-01-09 17:11:56,807 - WARNING - Found device ID not found on platform: linux
p1s_6-1  | 2025-01-09 17:11:57,205 - INFO - Handshake complete, server con wss://starport-v1.octoeverywhere.com/octoclientws[1], successfully connected to OctoEverywhere!
p1s_6-1  | 2025-01-09 17:11:57,206 - INFO - Primary Connection To OctoEverywhere Established - We Are Ready To Go!
p1s_6-1  | 2025-01-09 17:11:57,206 - WARNING - 
p1s_6-1  | 2025-01-09 17:11:57,206 - WARNING - 
p1s_6-1  | 2025-01-09 17:11:57,206 - WARNING - ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
p1s_6-1  | 2025-01-09 17:11:57,206 - WARNING - ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
p1s_6-1  | 2025-01-09 17:11:57,206 - WARNING -           This Plugin Isn't Connected To OctoEverywhere!          
p1s_6-1  | 2025-01-09 17:11:57,206 - WARNING -  Use the following link to finish the setup and get remote access:
p1s_6-1  | 2025-01-09 17:11:57,207 - WARNING -  https://octoeverywhere.com/getstarted?isFromKlipper=true&printerid=BZ2YA1X6NLIH5ZW9FS1CZYY9RQB5MI854W2E522A5D7MC94WB4Y4UKCTFA3P
p1s_6-1  | 2025-01-09 17:11:57,207 - WARNING - ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
p1s_6-1  | 2025-01-09 17:11:57,207 - WARNING - ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
p1s_6-1  | 2025-01-09 17:11:57,208 - WARNING - 
p1s_6-1  | 2025-01-09 17:11:57,208 - WARNING - 
p1s_6-1  | 2025-01-09 17:12:59,649 - WARNING - BambuWebcamHelper wasn't able to get the printer state after 6 attempts
p1s_6-1  | 2025-01-09 17:12:59,649 - INFO - QuickCam capture thread starting.
p1s_6-1  | 2025-01-09 17:13:03,900 - WARNING - BambuWebcamHelper wasn't able to get the printer state after 6 attempts
```

RyanEwen, 2025-01-09:
Setting env var CONNECTION_MODE=local has worked around the issue for now

QuinnDamerell, 2025-01-24:
Sorry for the delay. Yeah, at this time, Bambu has broken all of their cloud APIs for 3rd party developers, so the cloud option doesn't work anymore.  :(

Luckily, the local connection mode works and is frankly better since it's a more reliable connection. I updated all of the blog posts and docs to favor the local mode, so future users should see it first.

closed 2025-01-24 by QuinnDamerell