QuinnDamerell/OctoPrint-OctoEverywhere #89: Unable to set timezone in Docker

jwilson2899, 2025-01-09:
I would like to be able to set the timezone in docker so that logs and such in the container are marked with the correct timestamps. I tried doing it like I do with another container and specifying the following under the `environment:` section"

   `- TZ=America/New_York`

When I do that I don't get any errors, but the timezone remains set as UTC. In doing some digging I think the following needs to be added for this to work:

1.  `tzdata` package needs to be added to container if not already done
2.  /usr/share/zoneinfo directory needs to exists, with all of the timezone data directories under it
3.  /etc/timezone file needs to exist.

One of the timezone files (GMT) should be linked to /etc/localtime, and the /etc/timezone file should be populated with the value you specify in the env 

When I copy the /usr/share/zoneinfo directory from the other container, copy /usr/share/zoneinfo/America/New_York to /etc/localtime, and `echo "America/New_York" > /etc/timezone`, the date then changes properly the timezone and thus the timestamps are set properly in the container. As a further test, I added volumes to give me the /etc/timezone and /etc/localtime files, and /usr/share/zoneinfo direcotry with contents, and the date is now correct, and the OctoEverywhere logs show the proper time stamps. This can be used as a workaround for anyone who needs/wants this functionality, but if it could be natively included in the container image, that would be great.

QuinnDamerell, 2025-01-24:
Thanks for that. I will look into adding it to the docker container. If you are adventurous, you can also try it and send a PR!

jwilson2899, 2025-01-24:
> Thanks for that. I will look into adding it to the docker container. If you are adventurous, you can also try it and send a PR!

I’m starting to get more adventurous in the world of containers, so if I get some spare cycles over the next couple of days I’ll see if I can work something up. It should just be a small Dockerfile change and compose file change.

PR #93 (2025-01-26): Added tzdata to allow time zone to be set properly.
Added files to Container to allow for the time zone to be properly set in the compose file. These changes will allow the log files to show proper local time as long as the time zone is set properly in the compose file. It was defaulted to GMT, and ignoring it won't cause any issues, your logs will just be in GMT.

This PR Fixes #89
closed 2025-01-26 by QuinnDamerell