QuinnDamerell/OctoPrint-OctoEverywhere #90: Unable to start docker container

golden-moss, 2025-01-17:
The official docker image fails on startup due to a permissions issue. Running as sudo does not fix the issue, neither does manually creating the config file. Traceback below: 

```
2025-01-16 23:40:36,490 - INFO - ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2025-01-16 23:40:36,490 - INFO - Starting Docker OctoEverywhere Bootstrap
2025-01-16 23:40:36,490 - INFO - ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2025-01-16 23:40:36,491 - INFO - Env Vars: environ({'VENV_DIR': '/app/octoeverywhere-env', 'DATA_DIR': '/data/', 'PATH': '/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin', 'container': 'podman', 'ACCESS_CODE': '-', 'SERIAL_NUMBER': '-', 'REPO_DIR': '/app/octoeverywhere', 'USER': 'app', 'PRINTER_IP': '-', 'HOME': '/root', 'HOSTNAME': '-'})
2025-01-16 23:40:36,491 - INFO - Ensuring path exists: /app/octoeverywhere-env
2025-01-16 23:40:36,491 - INFO - Ensuring path exists: /app/octoeverywhere
2025-01-16 23:40:36,491 - INFO - Ensuring path exists: /data/
2025-01-16 23:40:36,491 - INFO - Init config object: /data/
Config file doesn't exist. Creating a new file now!
2025-01-16 23:40:36,492 - ERROR - Exception while bootstrapping up OctoEverywhere Bambu Connect.; PermissionError Exception: [Errno 13] Permission denied: '/data/octoeverywhere.conf'; Traceback (most recent call last):
  File "/app/octoeverywhere/docker_octoeverywhere/__main__.py", line 73, in <module>
    config = Config(configPath)
             ^^^^^^^^^^^^^^^^^^
  File "/app/octoeverywhere/linux_host/config.py", line 111, in __init__
    self._LoadConfigIfNeeded_UnderLock()
  File "/app/octoeverywhere/linux_host/config.py", line 308, in _LoadConfigIfNeeded_UnderLock
    self._SaveConfig_UnderLock()
  File "/app/octoeverywhere/linux_host/config.py", line 317, in _SaveConfig_UnderLock
    with open(self.OeConfigFilePath, 'w', encoding="utf-8") as f:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/data/octoeverywhere.conf'
```

QuinnDamerell, 2025-01-24:
That path /data/... is mapped to a directory outside the container for persistent storage. It's defined in your docker command line or the docker-compose file. Can you check that it's pointing to a path with the proper permissions?

golden-moss, 2025-01-24:
yes, the `~/octoeverywhere/data` folder is `755` and owned by the user running the container, and neither `chmod 777` or using the absolute path fix this problem. below is my `~/octoeverywhere/docker-compose.yml` with the env vars omitted:
```yml
version: '2'
services:
  octoeverywhere-bambu-connect:
    image: octoeverywhere/octoeverywhere:latest
    environment:
      - ACCESS_CODE=*
      - SERIAL_NUMBER=*
      - PRINTER_IP=*
    volumes:
      - ./data:/data
```

golden-moss, 2025-01-24:
nevermind lmao I messed up the SELinux config and that's what was stopping it, closing

closed 2025-01-24 by golden-moss