QuinnDamerell/OctoPrint-OctoEverywhere #92: bambu webcam lost, but can be reconnected after restarting octoservice

hotellonely, 2025-01-24:
On the webui, it says that the webcam feed is lost and cannot connect, but it still shows the current print status.

According to OctoEverywhere wiki, this is a bug from Bambu. However when I restarted the service using 

sudo systemctl restart octoeverywhere-bambu

which is basically to restart the service, then the video feed is back without the need of restarting the X1C.

This strongly hints that there's a problem within the service instead of the Bambu printer. We need an automatic restart of the service if this happens, as otherwise it's making the AI detection useless.

QuinnDamerell, 2025-01-24:
Hey! That's an interesting note; I have never seen restarting the plugin as a workaround. I have an X1 myself, and I will look into it. If a plugin restart works, that's exciting because then I might be able to fix the bug!

hotellonely, 2025-01-24:
It could be a different bug though, I was running it on a rpi-cm4 8gb ram model, I think the quickest thing we can try is to simply add a restart button on the webui and add the corresponding restart capability on the bambuconnect :)

closed 2025-05-15 by QuinnDamerell