QuinnDamerell/OctoPrint-OctoEverywhere #94: OctoPrint 1.11.0rc1 sentry version pinning

jneilliii, 2025-01-28:
OctoEverywhere has different pinned version than the new rc release. 

[image]

QuinnDamerell, 2025-01-28:
Thanks for letting me know! I’m not using sentry at all, so I can just disable it for now. @jneilliii , for the future, do you have any advice on the best way to handle PY packages that are both in OctoPrint and a plugin?

jneilliii, 2025-01-28:
If they are in OctoPrint already I tend to not add them to requirements or plugin_requires field in setup.py.

QuinnDamerell, 2025-01-29:
Good to know! Thanks!

QuinnDamerell, 2025-01-29:
This issue was fixed by https://github.com/QuinnDamerell/OctoPrint-OctoEverywhere/commit/24759466cbd06dfa0169aa8a9a2e2f797c50531e which was released in the update [3.6.8](https://github.com/QuinnDamerell/OctoPrint-OctoEverywhere/releases/tag/3.6.8).

Thanks again for letting me know!

closed 2025-01-29 by QuinnDamerell