QuinnDamerell/OctoPrint-OctoEverywhere #97: Can't print anymore

Ghimpe-ro, 2025-02-01:
After the last update my octoprint is not able to print. The print time is not available, the printer is slow and extrusion is half. 
The same GCODE file loaded on a card work flawlessly.

QuinnDamerell, 2025-02-01:
Do you recall what version you where on before the update? You might be able to find it in the log.

Ghimpe-ro, 2025-02-02:
Hi, there is a new update that did not solve anything. I usually keep it up to date when I use it and lately it was mostly active. 

Here is more info:
- a couple of pictures with a primitive (cilinder) printed like 5% just to see the difference. One is out of focus, but I can assure you, the  part printed from the printer's menu is sharp. The other one looks like it was melt in the sun for 10 years. 
- two recordings when printing the two samples. 
- a few images with the upgrade log, you can see some addons as well. 
- I noticed the temperature was lower than the GCODE by 10 deg C and the bed even lower, like 60 deg C less. Even with manual update of the temperature it was not working faster. 

Tell me what other data can I offer. 

https://github.com/user-attachments/assets/f393b075-36ba-462f-8b2d-9c7cf1f10dcb
https://github.com/user-attachments/assets/c153a86a-a657-4d86-974d-70501fed9a4d
[image]
[image]

[image]
[image]
[image]
[image]
[image]

Ghimpe-ro, 2025-02-05:
A few updates later it is solved. Here is the final log. 
[image]

closed 2025-02-05 by Ghimpe-ro