QuinnDamerell/OctoPrint-OctoEverywhere #108: Docker container crashes whenever I open app

jscherry, 2025-05-28:
I have deleted the container and started from scratch, I even reset the access code and created a new container with the new access code and it crashes every time I access the app?

docker compose:
version: '2'
services:
  octoeverywhere-bambu-connect:
    image: octoeverywhere/octoeverywhere:latest
    environment:
        # Requried to set the docker conntainer in Bambu Connect mode.
      - COMPANION_MODE=bambu
        # https://octoeverywhere.com/s/access-code
      - ACCESS_CODE=redacted
        # https://octoeverywhere.com/s/bambu-sn
      - SERIAL_NUMBER=redacted
        # https://octoeverywhere.com/s/bambu-ip
      - PRINTER_IP=redacted

    volumes:
      # This can also be an absolute path as well.
      - ./data:/data

logs:
2025-05-28 01:17:09,747 - INFO - ################################################

2025-05-28 01:17:09,747 - INFO - #### OctoEverywhere Bambu Connect Starting #####

2025-05-28 01:17:09,747 - INFO - ################################################

2025-05-28 01:17:09,748 - INFO - Plugin Version: 4.3.0

2025-05-28 01:17:09,753 - INFO - ZStandard Dict Training loaded. Data Length:112640 DictID:1

2025-05-28 01:17:09,753 - INFO - Compression is using zstandard with 2 threads

2025-05-28 01:17:09,754 - INFO - mDns Cache file loaded. Cached Entries Found: 0

2025-05-28 01:17:09,755 - INFO - OctoPingPong stats loaded from file.

2025-05-28 01:17:09,755 - INFO - PingPong has detected a first time run. Updating latency stats now.

2025-05-28 01:17:09,757 - INFO - Trying to connect to printer via local connection at 192.168.1.198...

2025-05-28 01:17:09,763 - INFO - Attempting to talk to OctoEverywhere, server con wss://starport-v1.octoeverywhere.com/octoclientws[1] wsId:140502724861088

2025-05-28 01:17:09,971 - INFO - Websocket connected

2025-05-28 01:17:09,971 - INFO - Connected To OctoEverywhere, server con wss://starport-v1.octoeverywhere.com/octoclientws[1]. Starting handshake...

2025-05-28 01:17:10,026 - WARNING - Found device ID not found on platform: linux

2025-05-28 01:17:10,100 - INFO - Handshake complete, server con wss://starport-v1.octoeverywhere.com/octoclientws[1], successfully connected to OctoEverywhere!

2025-05-28 01:17:10,101 - INFO - Primary Connection To OctoEverywhere Established - We Are Ready To Go!

2025-05-28 01:17:14,354 - INFO - Connection to the Bambu printer established! - Subscribing to the report subscription.

2025-05-28 01:17:14,461 - INFO - Starting full state sync.

2025-05-28 01:17:14,815 - ERROR - Sentry Error: Unknown printer type. CPU:BambuCPUs.ESP32, Project Name: None, Hardware Version: AP04

2025-05-28 01:17:14,815 - INFO - Printer Version: BambuPrinters.Unknown, CPU: BambuCPUs.ESP32, Project: None Hardware: AP04, Software: 01.08.01.00, Serial: 01D06B480901141

2025-05-28 01:17:15,813 - INFO - Print Manager recovered a print info from disk matching cookie: 340019893-215 Silver Parts_30

2025-05-28 01:17:15,813 - INFO - Restore client sync state: Restoring printing timer with existing duration of 1015.5571236610413

2025-05-28 01:17:15,814 - INFO - Gadget is now watching!

2025-05-28 01:17:17,814 - INFO - First layer timer is done. Stopping.

2025-05-28 01:17:17,814 - INFO - RepeatTimer thread exit

2025-05-28 01:17:35,640 - INFO - QuickCam capture thread starting.

2025-05-28 01:17:40,526 - ERROR - Exception in web stream [4] connect loop.; ValueError Exception: Buffer is empty; Traceback (most recent call last):

  File "/app/octoeverywhere/octoeverywhere/WebStream/octowebstream.py", line 131, in run

    self.mainThread()

  File "/app/octoeverywhere/octoeverywhere/WebStream/octowebstream.py", line 177, in mainThread

    returnValue = self.HttpHelper.IncomingServerMessage(webStreamMsg)

                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/app/octoeverywhere/octoeverywhere/WebStream/octowebstreamhttphelper.py", line 142, in IncomingServerMessage

    self.executeHttpRequest()

  File "/app/octoeverywhere/octoeverywhere/WebStream/octowebstreamhttphelper.py", line 393, in executeHttpRequest

    nonCompressedBodyReadSize, lastBodyReadLength, dataOffset = self.readContentFromBodyAndMakeDataVector(builderContext, octoHttpResult, boundaryStr, compressBody, contentTypeLower, contentLength, responseHandlerContext)

                                                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/app/octoeverywhere/octoeverywhere/WebStream/octowebstreamhttphelper.py", line 942, in readContentFromBodyAndMakeDataVector

    finalDataBuffer.Release()

  File "/app/octoeverywhere/octoeverywhere/buffer.py", line 136, in Release

    raise ValueError("Buffer is empty")

ValueError: Buffer is empty

2025-05-28 01:17:40,527 - ERROR - Session reported an error [wss://starport-v1.octoeverywhere.com/octoclientws[1]], closing the websocket. Backoff time sec: 1

2025-05-28 01:17:40,527 - INFO - Closing all open web stream sockets (1)

2025-05-28 01:17:40,527 - INFO - OctoServerCon websocket close start. IsPrimary?:True; wsId:140502724861088

2025-05-28 01:17:40,641 - INFO - OctoServerCon disconnect complete.

2025-05-28 01:17:45,840 - INFO - Creating new session for https://gadget-v1-oeapi.octoeverywhere.com

2025-05-28 01:17:46,689 - INFO - Gadget HostLock set to: gadget-ams.octoeverywhere.com

2025-05-28 01:17:59,669 - INFO - Service websocket closed.

2025-05-28 01:17:59,669 - INFO - Disconnected from OctoEverywhere, server con wss://starport-v1.octoeverywhere.com/octoclientws[1]

2025-05-28 01:17:59,670 - INFO - Closing all open web stream sockets (0)

2025-05-28 01:17:59,670 - INFO - Sleeping for 14 seconds before trying again.

2025-05-28 01:18:01,691 - INFO - Creating new session for https://gadget-ams.octoeverywhere.com

2025-05-28 01:18:13,671 - INFO - Attempting to talk to OctoEverywhere, server con wss://starport-v1.octoeverywhere.com/octoclientws[2] wsId:140502691436816

2025-05-28 01:18:15,209 - INFO - Websocket connected

2025-05-28 01:18:15,209 - INFO - Connected To OctoEverywhere, server con wss://starport-v1.octoeverywhere.com/octoclientws[2]. Starting handshake...

2025-05-28 01:18:15,223 - WARNING - Found device ID not found on platform: linux

2025-05-28 01:18:15,521 - INFO - Handshake complete, server con wss://starport-v1.octoeverywhere.com/octoclientws[2], successfully connected to OctoEverywhere!

2025-05-28 01:18:15,521 - INFO - Primary Connection To OctoEverywhere Established - We Are Ready To Go!

2025-05-28 01:18:42,159 - WARNING - Exception in QuickCam capture thread. The read operation timed out

QuinnDamerell, 2025-05-28:
Thank you for reporting this! I just pushed plugin version 4.3.1, which should fix this. The Docker image is building now. When it's ready, grab it and let me know if it fixes the issue!

closed 2025-05-29 by QuinnDamerell

jscherry, 2025-05-31:
still not working on 4.3.1. Went as far as deleting container again and starting from scratch. Verified serial and access code are correct.
yaml:
version: '2'
services:
  octoeverywhere-bambu-connect:
    image: octoeverywhere/octoeverywhere:latest
    environment:
        # Requried to set the docker conntainer in Bambu Connect mode.
      - COMPANION_MODE=bambu
        # https://octoeverywhere.com/s/access-code
      - ACCESS_CODE=redacted    # https://octoeverywhere.com/s/bambu-sn
      - SERIAL_NUMBER=redacted
        # https://octoeverywhere.com/s/bambu-ip
      - PRINTER_IP=192.168.1.198

    volumes:
      # This can also be an absolute path as well.
      - ./data:/data

logs:
2025-05-31 02:31:19,154 - WARNING - ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

2025-05-31 02:31:19,157 - INFO - Printer Version: BambuPrinters.Unknown, CPU: BambuCPUs.ESP32, Project: None Hardware: AP04, Software: 01.08.01.00, Serial: 01D06B480901141

2025-05-31 02:31:19,159 - WARNING - ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

2025-05-31 02:31:19,159 - WARNING - 

2025-05-31 02:31:19,159 - WARNING - 

2025-05-31 02:31:20,177 - INFO - Print Manager recovered a print info from disk matching cookie: 342883355-Red Cape Left and Right

2025-05-31 02:31:20,179 - INFO - Restore client sync state: Restoring printing timer with existing duration of 1214.302354335785

2025-05-31 02:31:20,182 - INFO - Gadget is now watching!

2025-05-31 02:31:22,183 - INFO - First layer timer is done. Stopping.

2025-05-31 02:31:22,183 - INFO - RepeatTimer thread exit

2025-05-31 02:31:50,463 - INFO - QuickCam capture thread starting.

2025-05-31 02:31:57,376 - INFO - Creating new session for https://gadget-v1-oeapi.octoeverywhere.com

2025-05-31 02:32:02,394 - INFO - Failed to send gadget inspection due to a connection error. HTTPSConnectionPool(host='gadget-v1-oeapi.octoeverywhere.com', port=443): Max retries exceeded with url: /api/gadget/inspect (Caused by NameResolutionError("<urllib3.connection.HTTPSConnection object at 0x727ad3538200>: Failed to resolve 'gadget-v1-oeapi.octoeverywhere.com' ([Errno -3] Try again)"))

2025-05-31 02:32:42,772 - INFO - Gadget HostLock set to: gadget-lon.octoeverywhere.com

2025-05-31 02:32:47,539 - WARNING - Exception in QuickCam capture thread. The read operation timed out

2025-05-31 02:32:53,134 - ERROR - Bambu printer mqtt connection lost when trying to sub for events.

2025-05-31 02:32:53,134 - ERROR - THIS USUALLY MEANS THE PRINTER SERIAL NUMBER IS WRONG. We tried to use the serial number '01P09C482401833'. Double check the SN is correct.

2025-05-31 02:32:53,135 - WARNING - Bambu printer connection lost. We will try to reconnect in a few seconds.

2025-05-31 02:32:53,135 - INFO - Sleeping for 0.0 seconds before trying to reconnect to the Bambu printer.

2025-05-31 02:32:53,139 - INFO - Trying to connect to printer via local connection at 192.168.1.198...

2025-05-31 02:32:56,467 - INFO - Connection to the Bambu printer established! - Subscribing to the report subscription.

2025-05-31 02:32:56,630 - INFO - Starting full state sync.

2025-05-31 02:32:57,119 - ERROR - Sentry Error: Unknown printer type. CPU:BambuCPUs.ESP32, Project Name: None, Hardware Version: AP04

2025-05-31 02:32:57,119 - INFO - Printer Version: BambuPrinters.Unknown, CPU: BambuCPUs.ESP32, Project: None Hardware: AP04, Software: 01.08.01.00, Serial: 01D06B480901141

2025-05-31 02:32:57,268 - INFO - Restore client sync state: We have the print cookie, detected an active print, and our timers are already running. So there's nothing to do.

2025-05-31 02:33:05,377 - WARNING - Exception in QuickCam capture thread. The read operation timed out

2025-05-31 02:33:09,970 - ERROR - Bambu printer mqtt connection lost when trying to sub for events.

2025-05-31 02:33:09,970 - ERROR - THIS USUALLY MEANS THE PRINTER SERIAL NUMBER IS WRONG. We tried to use the serial number 'redacted'. Double check the SN is correct.

2025-05-31 02:33:09,970 - WARNING - Bambu printer connection lost. We will try to reconnect in a few seconds.

2025-05-31 02:33:09,973 - INFO - Sleeping for 0.0 seconds before trying to reconnect to the Bambu printer.

2025-05-31 02:33:09,982 - INFO - Trying to connect to printer via local connection at 192.168.1.198...

2025-05-31 02:33:10,882 - INFO - Connection to the Bambu printer established! - Subscribing to the report subscription.

2025-05-31 02:33:10,889 - INFO - Starting full state sync.

2025-05-31 02:33:10,916 - ERROR - Sentry Error: Unknown printer type. CPU:BambuCPUs.ESP32, Project Name: None, Hardware Version: AP04

2025-05-31 02:33:10,916 - INFO - Printer Version: BambuPrinters.Unknown, CPU: BambuCPUs.ESP32, Project: None Hardware: AP04, Software: 01.08.01.00, Serial: 01D06B480901141

2025-05-31 02:33:10,980 - INFO - Restore client sync state: We have the print cookie, detected an active print, and our timers are already running. So there's nothing to do.

2025-05-31 02:34:00,759 - WARNING - Exception in QuickCam capture thread. The read operation timed out