QuinnDamerell/OctoPrint-OctoEverywhere #109: Docker container crashes whenever I open app

jscherry, 2025-05-31:
still not working on 4.3.1. Went as far as deleting container again and starting from scratch. Verified serial and access code are correct.
yaml:
version: '2'
services:
octoeverywhere-bambu-connect:
image: octoeverywhere/octoeverywhere:latest
environment:
# Requried to set the docker conntainer in Bambu Connect mode.
- COMPANION_MODE=bambu
# https://octoeverywhere.com/s/access-code
- ACCESS_CODE=redacted # https://octoeverywhere.com/s/bambu-sn
- SERIAL_NUMBER=redacted
# https://octoeverywhere.com/s/bambu-ip
- PRINTER_IP=192.168.1.198

volumes:
  # This can also be an absolute path as well.
  - ./data:/data
logs:
2025-05-31 02:31:19,154 - WARNING - ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

2025-05-31 02:31:19,157 - INFO - Printer Version: BambuPrinters.Unknown, CPU: BambuCPUs.ESP32, Project: None Hardware: AP04, Software: 01.08.01.00, Serial: 01D06B480901141

2025-05-31 02:31:19,159 - WARNING - ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

2025-05-31 02:31:19,159 - WARNING -

2025-05-31 02:31:19,159 - WARNING -

2025-05-31 02:31:20,177 - INFO - Print Manager recovered a print info from disk matching cookie: 342883355-Red Cape Left and Right

2025-05-31 02:31:20,179 - INFO - Restore client sync state: Restoring printing timer with existing duration of 1214.302354335785

2025-05-31 02:31:20,182 - INFO - Gadget is now watching!

2025-05-31 02:31:22,183 - INFO - First layer timer is done. Stopping.

2025-05-31 02:31:22,183 - INFO - RepeatTimer thread exit

2025-05-31 02:31:50,463 - INFO - QuickCam capture thread starting.

2025-05-31 02:31:57,376 - INFO - Creating new session for https://gadget-v1-oeapi.octoeverywhere.com/

2025-05-31 02:32:02,394 - INFO - Failed to send gadget inspection due to a connection error. HTTPSConnectionPool(host='gadget-v1-oeapi.octoeverywhere.com', port=443): Max retries exceeded with url: /api/gadget/inspect (Caused by NameResolutionError("<urllib3.connection.HTTPSConnection object at 0x727ad3538200>: Failed to resolve 'gadget-v1-oeapi.octoeverywhere.com' ([Errno -3] Try again)"))

2025-05-31 02:32:42,772 - INFO - Gadget HostLock set to: gadget-lon.octoeverywhere.com

2025-05-31 02:32:47,539 - WARNING - Exception in QuickCam capture thread. The read operation timed out

2025-05-31 02:32:53,134 - ERROR - Bambu printer mqtt connection lost when trying to sub for events.

2025-05-31 02:32:53,134 - ERROR - THIS USUALLY MEANS THE PRINTER SERIAL NUMBER IS WRONG. We tried to use the serial number '01P09C482401833'. Double check the SN is correct.

2025-05-31 02:32:53,135 - WARNING - Bambu printer connection lost. We will try to reconnect in a few seconds.

2025-05-31 02:32:53,135 - INFO - Sleeping for 0.0 seconds before trying to reconnect to the Bambu printer.

2025-05-31 02:32:53,139 - INFO - Trying to connect to printer via local connection at 192.168.1.198...

2025-05-31 02:32:56,467 - INFO - Connection to the Bambu printer established! - Subscribing to the report subscription.

2025-05-31 02:32:56,630 - INFO - Starting full state sync.

2025-05-31 02:32:57,119 - ERROR - Sentry Error: Unknown printer type. CPU:BambuCPUs.ESP32, Project Name: None, Hardware Version: AP04

2025-05-31 02:32:57,119 - INFO - Printer Version: BambuPrinters.Unknown, CPU: BambuCPUs.ESP32, Project: None Hardware: AP04, Software: 01.08.01.00, Serial: 01D06B480901141

2025-05-31 02:32:57,268 - INFO - Restore client sync state: We have the print cookie, detected an active print, and our timers are already running. So there's nothing to do.

2025-05-31 02:33:05,377 - WARNING - Exception in QuickCam capture thread. The read operation timed out

2025-05-31 02:33:09,970 - ERROR - Bambu printer mqtt connection lost when trying to sub for events.

2025-05-31 02:33:09,970 - ERROR - THIS USUALLY MEANS THE PRINTER SERIAL NUMBER IS WRONG. We tried to use the serial number 'redacted'. Double check the SN is correct.

2025-05-31 02:33:09,970 - WARNING - Bambu printer connection lost. We will try to reconnect in a few seconds.

2025-05-31 02:33:09,973 - INFO - Sleeping for 0.0 seconds before trying to reconnect to the Bambu printer.

2025-05-31 02:33:09,982 - INFO - Trying to connect to printer via local connection at 192.168.1.198...

2025-05-31 02:33:10,882 - INFO - Connection to the Bambu printer established! - Subscribing to the report subscription.

2025-05-31 02:33:10,889 - INFO - Starting full state sync.

2025-05-31 02:33:10,916 - ERROR - Sentry Error: Unknown printer type. CPU:BambuCPUs.ESP32, Project Name: None, Hardware Version: AP04

2025-05-31 02:33:10,916 - INFO - Printer Version: BambuPrinters.Unknown, CPU: BambuCPUs.ESP32, Project: None Hardware: AP04, Software: 01.08.01.00, Serial: 01D06B480901141

2025-05-31 02:33:10,980 - INFO - Restore client sync state: We have the print cookie, detected an active print, and our timers are already running. So there's nothing to do.

2025-05-31 02:34:00,759 - WARNING - Exception in QuickCam capture thread. The read operation timed out

jscherry, 2025-05-31:
I had this running well for the past several months and now I'm continually stuck in a reboot loop or something?

current logs:
2025-05-31 22:25:44,691 - INFO - Trying to connect to printer via local connection at 192.168.1.198...

2025-05-31 22:25:46,612 - INFO - Connection to the Bambu printer established! - Subscribing to the report subscription.

2025-05-31 22:25:46,619 - INFO - Starting full state sync.

2025-05-31 22:25:46,635 - ERROR - Sentry Error: Unknown printer type. CPU:BambuCPUs.ESP32, Project Name: None, Hardware Version: AP04

2025-05-31 22:25:46,636 - INFO - Printer Version: BambuPrinters.Unknown, CPU: BambuCPUs.ESP32, Project: None Hardware: AP04, Software: 01.08.01.00, Serial: redacted
2025-05-31 22:25:46,723 - INFO - Restore client sync state: There's no print and none of the timers are running.

2025-05-31 22:26:00,763 - ERROR - Bambu printer mqtt connection lost when trying to sub for events.

2025-05-31 22:26:00,763 - ERROR - THIS USUALLY MEANS THE PRINTER SERIAL NUMBER IS WRONG. We tried to use the serial number 'redacted'. Double check the SN is correct.

2025-05-31 22:26:00,763 - WARNING - Bambu printer connection lost. We will try to reconnect in a few seconds.

2025-05-31 22:26:00,763 - INFO - Sleeping for 0.0 seconds before trying to reconnect to the Bambu printer.

2025-05-31 22:26:00,774 - INFO - Trying to connect to printer via local connection at 192.168.1.198...

2025-05-31 22:26:05,858 - WARNING - Failed to connect to the Bambu printer 192.168.1.198:8883, we will retry in a bit. timed out

2025-05-31 22:26:05,859 - INFO - Sleeping for 5.0 seconds before trying to reconnect to the Bambu printer.

2025-05-31 22:26:10,862 - INFO - Trying to connect to printer via local connection at 192.168.1.198...

2025-05-31 22:26:15,957 - WARNING - Failed to connect to the Bambu printer 192.168.1.198:8883, we will retry in a bit. timed out

2025-05-31 22:26:16,487 - INFO - Sleeping for 10.0 seconds before trying to reconnect to the Bambu printer.

2025-05-31 22:26:26,552 - INFO - Trying to connect to printer via local connection at 192.168.1.198...

2025-05-31 22:26:32,047 - WARNING - Failed to connect to the Bambu printer 192.168.1.198:8883, we will retry in a bit. timed out

2025-05-31 22:26:38,766 - INFO - Sleeping for 15.0 seconds before trying to reconnect to the Bambu printer.

2025-05-31 22:26:53,823 - INFO - Trying to connect to printer via local connection at 192.168.1.198...

2025-05-31 22:26:59,592 - INFO - Connection to the Bambu printer established! - Subscribing to the report subscription.

2025-05-31 22:26:59,602 - INFO - Starting full state sync.

2025-05-31 22:26:59,617 - ERROR - Sentry Error: Unknown printer type. CPU:BambuCPUs.ESP32, Project Name: None, Hardware Version: AP04

2025-05-31 22:26:59,617 - INFO - Printer Version: BambuPrinters.Unknown, CPU: BambuCPUs.ESP32, Project: None Hardware: AP04, Software: 01.08.01.00, Serial: redacted

2025-05-31 22:27:00,413 - INFO - Restore client sync state: There's no print and none of the timers are running.

2025-05-31 22:27:21,949 - ERROR - Bambu printer mqtt connection lost when trying to sub for events.

2025-05-31 22:27:21,953 - ERROR - THIS USUALLY MEANS THE PRINTER SERIAL NUMBER IS WRONG. We tried to use the serial number 'redacted'. Double check the SN is correct.

2025-05-31 22:27:21,954 - WARNING - Bambu printer connection lost. We will try to reconnect in a few seconds.

2025-05-31 22:27:21,954 - INFO - Sleeping for 0.0 seconds before trying to reconnect to the Bambu printer.

2025-05-31 22:27:21,960 - INFO - Trying to connect to printer via local connection at 192.168.1.198...

2025-05-31 22:27:27,512 - WARNING - Failed to connect to the Bambu printer 192.168.1.198:8883, we will retry in a bit. timed out

2025-05-31 22:27:27,512 - INFO - Sleeping for 5.0 seconds before trying to reconnect to the Bambu printer.

2025-05-31 22:27:32,515 - INFO - Trying to connect to printer via local connection at 192.168.1.198...

2025-05-31 22:27:38,474 - WARNING - Failed to connect to the Bambu printer 192.168.1.198:8883, we will retry in a bit. timed out

[... 104 lines cut ...]

2025-05-31 22:30:18,696 - ERROR - Sentry Error: Unknown printer type. CPU:BambuCPUs.ESP32, Project Name: None, Hardware Version: AP04

2025-05-31 22:30:18,697 - INFO - Printer Version: BambuPrinters.Unknown, CPU: BambuCPUs.ESP32, Project: None Hardware: AP04, Software: 01.08.01.00, Serial: redacted
2025-05-31 22:30:18,783 - INFO - Restore client sync state: There's no print and none of the timers are running.

2025-05-31 22:31:02,360 - ERROR - Bambu printer mqtt connection lost when trying to sub for events.

2025-05-31 22:31:02,361 - ERROR - THIS USUALLY MEANS THE PRINTER SERIAL NUMBER IS WRONG. We tried to use the serial number 'redacted'. Double check the SN is correct.

2025-05-31 22:31:02,362 - WARNING - Bambu printer connection lost. We will try to reconnect in a few seconds.

2025-05-31 22:31:02,362 - INFO - Sleeping for 0.0 seconds before trying to reconnect to the Bambu printer.

2025-05-31 22:31:02,365 - INFO - Trying to connect to printer via local connection at 192.168.1.198...

2025-05-31 22:31:04,625 - INFO - Connection to the Bambu printer established! - Subscribing to the report subscription.

2025-05-31 22:31:04,635 - INFO - Starting full state sync.

2025-05-31 22:31:04,656 - ERROR - Sentry Error: Unknown printer type. CPU:BambuCPUs.ESP32, Project Name: None, Hardware Version: AP04

2025-05-31 22:31:04,656 - INFO - Printer Version: BambuPrinters.Unknown, CPU: BambuCPUs.ESP32, Project: None Hardware: AP04, Software: 01.08.01.00, Serial: redacted

2025-05-31 22:31:04,717 - INFO - Restore client sync state: There's no print and none of the timers are running.

QuinnDamerell, 2025-07-05:
Thanks for opening this! From the most recent log, it looks like there might be a serial number mismatch. Can you check the serial it's using matches the printer, and the printer's IP hasn't changed?

closed 2025-08-02 by QuinnDamerell